rusefi/rusefi #714: we need TLE8888 driver

rusefi, 2019-03-25:
First step would be to execute Output Enable command

rusefi, 2019-04-12:
See also https://github.com/dron0gus/rusefi/blob/board-subaru/firmware/config/boards/subaruej20g/board.c#L91

rusefi, 2019-04-13:
Next chapter is to get smart outputs working. At the moment smart pins are not working

@dron0gus 

[image]

rusefi, 2019-04-14:
Still failing to get 
https://github.com/rusefi/rusefi/blob/master/firmware/config/engines/custom_engine.cpp#L369
to work as expected on a bench

rusefi, 2019-04-14:
A bit better status
[image]

and TLE_20 smart output works now!!!

dron0gus, 2019-04-14:
FYI: current implementation has a lot of limitations.
Most annoying for now is that driver can not handle TLE8888_PIN_1..4 and TLE8888_PIN_24..27 (direct driven INJ and IGN channels).
Driver knows these are direct drive outputs but has no knowledge which STM pins should be used to control these outputs.
So please use STM native ports when configuring IGN and INJ channels.
Later we should switch to TLE8888_PIN_* if we want to map diagnostic information from chip to smart gpios...

Also we should add TLE8888_PIN_24..27 to enum and extend TLE8888_OUTPUTS to 28.

rusefi, 2019-04-15:
see #697

rusefi, 2019-05-11:
direct outputs work, smart pins work - good enough for now :)

closed 2019-05-11 by rusefi