rusefi/rusefi #718: stepper driver: some pin modes are hard-coded

rusefi, 2019-03-29:
while we have stepperEnablePinMode setting, the code is not using it!

also

andreika-git, 2019-11-24:
#1029

rusefillc, 2022-09-16:
```
void StepDirectionStepper::initialize(brain_pin_e stepPin, brain_pin_e directionPin, pin_output_mode_e directionPinMode, float reactionTime, brain_pin_e enablePin, pin_output_mode_e enablePinMode) {
	if (!isBrainPinValid(stepPin) || !isBrainPinValid(directionPin)) {
		return;
	}

	setReactionTime(reactionTime);

	this->directionPinMode = directionPinMode;
	this->directionPin.initPin("Stepper DIR", directionPin, &this->directionPinMode);

	this->stepPinMode = OM_DEFAULT;	// todo: do we need configurable stepPinMode?
	this->stepPin.initPin("Stepper step", stepPin, &this->stepPinMode);

	this->enablePinMode = enablePinMode;
	this->enablePin.initPin("Stepper EN", enablePin, &this->enablePinMode);
```

looks to be taken care of

closed 2022-09-16 by rusefillc