rusefi/rusefi #721: generate_memory_usage_report/gcc_map_reader BUG?

andreika-git, 2019-03-30:
Current version of java_tools/gcc_map_reader shows only partial RAM usase.
Example MAP file:
[rusefi.map.zip](https://github.com/rusefi/rusefi/files/3025289/rusefi.map.zip)
The output of gcc_map_reader:
[buggy_ram_report.txt](https://github.com/rusefi/rusefi/files/3025292/buggy_ram_report.txt)
The complete (correct) RAM usage report generated by "AMAP" software:
[correct_ram_report.txt](https://github.com/rusefi/rusefi/files/3025294/correct_ram_report.txt)
The first problem is gcc_map_reader doesn't process .data section (DATA_RAM), only .bss section is processed.
The second problem is gcc_map_reader doesn't even parse ALL .bss fields! For example, this one is missing: .bss.ch_idle_thread_wa (1552 bytes).

rusefi, 2019-03-30:
Just fixed the second problem - missed ch_idle_thread_wa

rusefi, 2019-03-31:
Good enough now?

rusefi, 2019-03-31:
В самом начале .map файла идут "Discarded input sections", т.е. всё, что там перечислено, gcc решил в целях оптимизации не размещать в памяти. У них всех нулевые адреса. Их нужно игнорить. А "правильные" поля идут уже ниже, после "Linker script and memory map"

rusefi, 2019-04-05:
@andreika-git is this good enough to close ticket?

andreika-git, 2019-04-05:
Yes!

commit d973ebaa (2019-03-30): generate_memory_usage_report/gcc_map_reader BUG? fix #721
closed 2019-04-05 by rusefi