rusefi/rusefi #727: gen_config: add rusefi_config_custom.txt to override #defines

andreika-git, 2019-04-01:
There is an idea to override custom config #defines for integration/rusefi_config.txt
It will allow more flexibility to TS configuration files (change size of some tables or skip some fields)
One of the ways is to:
1) parse rusefi_config_custom.txt before rusefi_config.txt (if exists)
2) ignore multiple #defines so that only the first one is used.
3) make sure that zero-sized arrays are processed correctly.

rusefi, 2019-04-03:
@andreika-git I believe it's done - simply add 5th parameter path to "custom.txt" file with same #define syntax

rusefi, 2019-04-05:
@andreika-git is this good enough to close ticket?

andreika-git, 2019-04-05:
Maybe it would be better to use separate configs for different boards? Currently we have common tunerstudio/rusefi.ini and controllers/algo/engine_configuration_generated_structures.h etc., and if I call gen_config with non-default parameters - they will be overwritten (and git will be angry). So maybe we should use the folder structure similar to config/boards ? 
One of the gen_config's arguments is "tunerstudio" - and it's not good because it's shared for input (rusefi.input) - which is common for all configs, and output (rusefi.ini) - which may vary per config.
If you think that it's irrelevant to this issue, then Yes, it's good enough to close.

closed 2025-07-19 by rusefillc