rusefi/rusefi #764: USB not working for F4 or F767

960, 2019-04-18:
With current configuration, I can not get USB to connect, and not TLE8888 to work.

I did this to get USB back working, but still no TLE8888 working.

#define BOARD_TLE6240_COUNT         0

rusefi, 2019-04-19:
@960 yes, there is an issue here. trying to fix it.

rusefi, 2019-04-19:
problem with PD8 pin used by joystick.

No idea how is this related to SPI pins so far

[image]

rusefi, 2019-04-19:
With some changes now we are at

[image]

rusefi, 2019-04-19:
@dron0gus finally getting to something interesting

[image]

960, 2019-04-20:
Have you found why TLE8888 isn't working?

rusefi, 2019-04-21:
It's a bit of a hot mess - turned out that ICU/PAL migration had issues, also continues integration had issues :( a bit of everything. in the end default firmware now has 8888 enabled

rusefi, 2019-04-21:
@960 I believe this is now resolved
TLE8888 is now enabled in default firmware

closed 2019-04-21 by rusefi