rusefi/rusefi #773: All corrections should be ratio, 1.0 = no correction

OrchardPerformance, 2019-04-23:
"IAT fuel", "Cranking TPS" and "Cranking CLT" are decimal multipliers while "Warmup Fuel Manual multiplier" and "Warmup Idle multiplier" are % values.

% values need changing to ratio so that "no correction" is always "1.0"

rusefi, 2019-04-23:
TODO at least one curve displays 100 instead of 1 for ratio

[image]

FYI @mck1117

rusefi, 2019-05-10:
@mck1117 poke :)

rusefi, 2019-05-31:
See also #811

rusefi, 2019-06-12:
@mck1117 I would like to state for the record that a month ago I've poked and I was hoping to get at least _some_ response within a month.

rusefi, 2019-06-13:
Thank you @mck1117  for the patience, great change!

mck1117, 2019-07-06:
I think this can close

closed 2019-07-06 by rusefi

rusefi, 2019-07-07:
@mck1117 says
```
The conclusion with ratio vs. % was that the generator tool needs to be able to emit things like:
```#if USE_PERCENT
    cltFuelCorr                        = array, F32,    10336,    [16],    "%", 100, 0, 0.0, 500.0, 0
#else
    cltFuelCorr                        = array, F32,    10336,    [16],    "ratio", 1, 0, 0.0, 5.0, 2
#endif```
```