rusefi/rusefi #775: electrical noise during tuning ETB causes full tune reset

rusefi, 2019-04-25:
While tuning ETB with @mck1117 we have experienced multiple ECU reboots during "Burn" commands

These reboots have caused tune to be lost which is not expected behavior. Expected behavior is that only one cope of the tune is written to flash at each given moment so that in case of electrical reboot we still have second tune copy.

rusefi, 2019-04-25:
See also https://sourceforge.net/p/rusefi/tickets/335/

rusefi, 2019-04-25:
the issue is about banks of RAM erases or written completely. Those banks are NOT 4k as in
```
/**
 * https://sourceforge.net/p/rusefi/tickets/335/
 *
 * Address of second config copy, rounded to 4K. 4K is the page size is it?
 *
 */
#define FLASH_ADDR_SECOND_COPY (FLASH_ADDR + ((PERSISTENT_SIZE + 4095) & 0xFFFFF000))

```

In reality those are way bigger
```
size_t flashSectorSize(flashsector_t sector) {
	// sectors 0..11 are the 1st memory bank (1Mb), and 12..23 are the 2nd (the same structure).
	if (sector <= 3 || (sector >= 12 && sector <= 15))
		return 16 * 1024;
	else if (sector == 4 || sector == 16)
		return 64 * 1024;
	else if ((sector >= 5 && sector <= 11) || (sector >= 17 && sector <= 23))
		return 128 * 1024;
	return 0;
}
```

rusefi, 2019-04-25:
[image]

mck1117, 2019-04-25:
So should each copy of the tune use a full 128k sector?

rusefi, 2019-04-25:
@mck1117 yes
at the moment we use sector #11 

```
#ifndef FLASH_ADDR
#define FLASH_ADDR 0x080E0000
#endif
```

mck1117, 2019-04-25:
Both copies, or one copy?

rusefi, 2019-04-25:
prior to the commit above both copies in sector 11

now with https://github.com/rusefi/rusefi/commit/14831e1da82cc02fa79a23c92d9814303515c16c two copies go into different sectors

closed 2019-04-26 by rusefi

rusefi, 2019-04-26:
https://www.youtube.com/watch?v=R9Vt3tQfPXE

commit c6c47392 (2019-04-26): electrical noise during tuning ETB causes full tune reset fix #775