rusefi/rusefi #778: Injection/Ignition angle inaccuracy on 60-2? EngineSniffer vs Real Hardware

andreika-git, 2019-04-28:
Test settings:
- trigger 60-2, trigger-offset = 0
- advance = 0, no corrections
Engine Sniffer seems ok:
[image]
But the real signals are shifted (~3.9 degrees?):
[image]
This delta seems to be constant for the given config.

[8 MHz, 480 M Samples [3].logicdata.zip](https://github.com/rusefi/rusefi/files/3125493/8.MHz.480.M.Samples.3.logicdata.zip)

mck1117, 2019-04-29:
Debug build, or release build? Internal trigger simulator, or something else generating it?

rusefi, 2019-04-29:
@mck1117 I know it's an external arduino stimulator. I would guess custom release build since Prometheus hardware with 469 CPU probably

rusefi, 2019-04-29:
I would like to use this as another change to say that 407 has to be the shared platform where most confusing issues have to be worked on, otherwise we have too much variation in hardware errdada configurations etc that it's impossible to investigate :(

andreika-git, 2019-05-06:
More data FYI. Two trigger types tested (60-2 and 36-1), on both Debug and Release firmware versions:
[image]
[image]
[image]
[image]

andreika-git, 2019-05-06:
The average shift in angles:
60-2-DEBUG 4.54°
60-2-RELEASE 3.93°
36-1-DEBUG 6.51°
36-1-RELEASE 5.87°
Analyzer signals attached.
[60-2-and-36-1.zip](https://github.com/rusefi/rusefi/files/3147756/60-2-and-36-1.zip)

andreika-git, 2019-05-06:
One-tooth trigger has the same issue.
But on a low RPM (~120) it has additional problems with dwell (constant dwell 4.0 ms in settings doesn't really work):
[test 1-tooth RPM=120 #2.logicdata.zip](https://github.com/rusefi/rusefi/files/3148373/test.1-tooth.RPM.120.2.logicdata.zip)
[image]

mck1117, 2019-05-06:
And this is on an f469?

andreika-git, 2019-05-06:
Sure

mck1117, 2019-05-06:
Can you share the tunerstudio project for this config?  I'd like to try and repro on an f407 to compare.

andreika-git, 2019-05-06:
Here it is:
[CurrentTune.msq.zip](https://github.com/rusefi/rusefi/files/3149364/CurrentTune.msq.zip)
mainController.ini is from current (latest) version. Change the trigger type if needed (currently it's "one-tooth"), and don't forget to change hw-pins config.

rusefi, 2019-05-06:
I have a theory about dwell issue:
trigger input is very uneven (high fronts duration goes from 0.21s to 0.23s for example) so RPM is uneven so time-based prediction of dwell start time would not be exactly 4ms before the dwell end time which is driven my trigger event. In most extreme cases it causes complete skip of spark! interesting effect an open question how to handle it, but this starts to look like a separate from original issue

Does it sound realistic? We need some measure of signal quality and we need to agree on what is expected what is not. I am sure that what this Ardino is producing is quite realistic. The benefit of not accurate Arduino is that it shows us issues which are not easily noticeable with more precise stm32 simulated signal

rusefi, 2019-05-07:
Slack room https://rusefi.slack.com/messages/CJ4HBS942/details/

rusefi, 2019-05-07:
https://www.youtube.com/watch?v=2qMylZ1t1jk

mck1117, 2019-05-08:
I got it to repro.  24-1 trigger signal.

Attached is a zip that contains full tunerstudio project, binaries flashed to device, and a full dump of the f407's flash.

[repro 778.zip](https://github.com/rusefi/rusefi/files/3156422/repro.778.zip)

andreika-git, 2019-05-08:
I've reproduced the "missing dwell" issue on 24-1 trigger signal, RPM=60.
[image]
[image]
[24-1-60rpm-missing-sparks.msl.zip](https://github.com/rusefi/rusefi/files/3156572/24-1-60rpm-missing-sparks.msl.zip)
[missing_dwell_rpm=60_24-1_trigger.logicdata.zip](https://github.com/rusefi/rusefi/files/3156573/missing_dwell_rpm.60_24-1_trigger.logicdata.zip)

andreika-git, 2019-05-08:
Some isolated dwell-missing cases are noticable even on rpm=140 (24-1 trigger):
[image]

[missing_dwell_140rpm_msl_and_logicdata.zip](https://github.com/rusefi/rusefi/files/3156649/missing_dwell_140rpm_msl_and_logicdata.zip)

andreika-git, 2019-05-10:
[image]

rusefi, 2019-05-10:
Missing dwell extracted to #796

rusefi, 2019-05-10:
status update: there is a chance that this is all about VR polarity. Open question is if rusEfi can detect flipped VR polarity and how.

I do not have a VR in my life. I am looking to have engine sniffer recording of flipped and not flipped VR sensor. 

without "handle only front" flag in rusEfi so that both fall and rises are recoded

rusefi, 2019-05-11:
See also #659

rusefi, 2019-11-06:
Looks like May 6th theory was pretty close to explain #796 issue

here we probably have something different?

@andreika-git can you still reproduce this? Shall we give this issue another try?

andreika-git, 2019-11-19:
Steps to reproduce:
1) Trigger_type = 60-2, Trigger_offset = 0
2) Timing_Mode = "fixed" (1), Fixed_Timing = 0
3) rpm = 200

Here's what we have now with virtual simulator:
[image]
The Virtual simulator works fine (tested on snapshot_20191119_001807_stm32f407_rusefi).
But on the real hardware (Deucalion + arduino-sim) I still have the same wrong offset as before:
[image]
[image]
So the issue is still there...

mck1117, 2022-06-22:
I'm pretty sure this is long since fixed