rusefi/rusefi #779: More CAN messages for VAG

rusefi, 2019-04-30:
@960 is asking for RPM and CLT CAN messages, this is somehow related to his cooling fan?

See also https://hackaday.io/project/6288/instructions

See also https://github.com/ElDominio/CANforVAGME75/blob/master/CANforVAGME75.ino

rusefi, 2019-04-30:
@960 FYI https://github.com/ElDominio/CANforVAGME75/blob/master/CANforVAGME75.ino is not great since has ZERO explanation for messages format, just posts magic bytes - how do I know which one is which?

rusefi, 2019-04-30:
@960 https://hackaday.io/project/6288/instructions has signal_watertemp  only as part of engine_control which is used as part of 'Diesel engine' only?

//Diesel engine
    CanSend(0x480, 0x00, engine_control, 0x00, 0x00, 0x00, temp_dpf_warning, 0x00, 0x00);

rusefi, 2019-04-30:
@960 better documentation is needed. these two links are not exactly great :(

Also rusEfi already seems to be sending VAG CLT?

```
	commonTxInit(CAN_VAG_CLT);
	setShortValue(&txmsg, (int) ((engine->sensors.clt + 48.373) / 0.75), 1); //Coolant Temp
	sendCanMessage();
```

960, 2019-04-30:
It seems adress for temperature are 0x288 10MS

It's byte nr 2       startbimbyte 0,     frame 8                             -48 / 142.5 ◦ C |(PH)=0.75*(HEX)-48 |oder Ersatzwert

I have can input and output from board, but still not working.

void CanSend(short address, byte a, byte b, byte c, byte d, byte e, byte f, byte g, byte h)
{
  unsigned char DataToSend[8] = {a,b,c,d,e,f,g,h};
  CAN.sendMsgBuf(address, 0, 8, DataToSend);
}

rusefi, 2019-04-30:
rusEfi current has
#define CAN_VAG_CLT 0x289

can you try changing to 0x288 maybe there is a typo?

960, 2019-04-30:
I have already tried.

But I dont have RPM either

static void canDashboardVAG(void) {
	//VAG Dashboard
	commonTxInit(CAN_VAG_RPM);
	setShortValue(&txmsg, GET_RPM() * 4, 2); //RPM
	sendCanMessage();

	commonTxInit(CAN_VAG_CLT);
	setShortValue(&txmsg, (int) ((engine->sensors.clt + 48.373) / 0.75), 1); //Coolant Temp
	sendCanMessage();
}

This is the code for VAG, it seems to be missing somethings.

How can that give a output like this:

0x280, 0, 8, 0x09, 0x22, rpmLo, rpmHi, 0x20, 0x00, 0x22, 0x22

rusefi, 2019-04-30:
@960 I am sorry I do not understand you. What is 0x22? I have no idea what you are talking about. these values make no sense to me at all :( Please provide more details.

Please attach you current tune I would like to confirm that CAN was configured properly

please run rusEfi console and execute "caninfo" command and post output here.

960, 2019-04-30:
I'll look a little more into the exact messages.

To explain the example above, forget the other values.

RPM are byte no 3 and 4.

This is the exact frame format to get rpm working.

0x280, 0, 8, 0x00, 0x20, rpmLo, rpmHi, 0x00, 0x00, 0x00, 0x00

960, 2019-04-30:
https://wiki.openstreetmap.org/wiki/VW-CAN

rusefi, 2019-04-30:
@960 i wound not chat using email or github. if you would like to chat please use slack or skype or any other chat platform

i believe the code does exactly what you say. if you are trying to clarify, i am not getting it. i need the information i have requested, i have nothing to add until i get it

960, 2019-05-01:
Finally working!

Need to set can nbc type manually in tune with notepad++

The setting are not enabled in TS

rusefi, 2019-05-01:
Just added canNbcType to TS. Weird why would it hidden there, I wonder what was my thinking :(

Thank you for the help, we've fixed on of the codes here right?

closed 2019-05-01 by rusefi

960, 2019-05-01:
Just tested with CAN-Sniffer that the caNbcType settings in TS are working. :-)

Thanks!

commit 55cf1083 (2019-05-01): More CAN messages for VAG fix #779