rusefi/rusefi #785: Missing ADC Pins for ADC3, PF port

960, 2019-05-04:
Pins for 144 pin mcu, ADC3 missing in both code and TS

PF3 to PF10

rusefi, 2019-05-04:
multiple issues here actually
1) ADC channels are hard-coded
```
// is there a reason to have this configurable?
#ifndef ADC_SLOW_DEVICE
#define ADC_SLOW_DEVICE ADCD1
#endif /* ADC_SLOW_DEVICE */

// is there a reason to have this configurable?
#ifndef ADC_FAST_DEVICE
#define ADC_FAST_DEVICE ADCD2
#endif /* ADC_FAST_DEVICE */
```
2) even for hard-coded ADC channels we sometimes have two options for ADC input pin but rusEfi has only one option hard coded

First step of the change would probably be to switch from using "input channels" in configuration to using "brain pins" for input configuration

rusefi, 2019-05-14:
need custom compile with
-DSTM32_ADC_USE_ADC3=TRUE -DADC_SLOW_DEVICE=ADCD3 -DEFI_PIN_ADC9=GPIOF_3

maybe this would make channel9 PF3 work for CLT and IAT? if that would work we can go from there

See https://www.st.com/resource/en/datasheet/stm32f769ni.pdf

mck1117, 2021-03-19:
duplicate of #2238

closed 2021-03-19 by mck1117

rusefillc, 2024-11-08:
See also https://github.com/rusefi/rusefi/issues/3350