rusefi/rusefi #801: Kg/H airflow gauge displays incorrect value

OrchardPerformance, 2019-05-19:
Tuner studio "Air Flow" gauge reads roughly half of the value in the MAF transfer table. This is consistent throughout the range of the MAF. 

[image]

OrchardPerformance, 2019-05-19:
10kg/h on the table reads 6.1kg/h on the gauge
100kg/h on the table reads 56.4kg/h on the gauge
500kg/h on the table reads 250.6kg/h on the gauge

So it looks like the MAF kg/h value is being halved somewhere.

mck1117, 2019-06-07:
Can I propose we also switch to g/s instead of kg/hr? SI base units are much preferable to hours.

OrchardPerformance, 2019-07-08:
kg/h is industry standard for air flow, the majority of MAF transfer functions will be in kg/h. 
It also helps hide any inaccuracy of the measurement values the user inputs, being 0.1kg/h out is only 0.027g/s but 0.1g/s is 0.36kg/h so rounding or interpolation errors are exaggerated.

mck1117, 2019-07-08:
> so rounding or interpolation errors are exaggerated

That's not necessarily the case - we use floating point internally, so we get ~9 digits of decimal precision.  No rounding happens until the very end when we convert to a pulse width, and even then, that's in microseconds, so we still get 4 or 5 significant digits.

That said - it does help the user not screw it up by getting the conversion wrong if they have kg/h handy already.

OrchardPerformance, 2019-07-08:
I was thinking more the kind of rounding a user would do when copying out of an excel sheet or typing it in badly. :) 

I agree if its going to already be in kg/h from their spec sheet or OEM calibration then that's a good reason not to alter it.

OrchardPerformance, 2019-07-24:
Things appear to be working since Matt's fix, I have tested it in my MAF branch and its  reading perfectly now.

closed 2019-07-24 by OrchardPerformance

mck1117, 2019-07-24:
For sake of accurate record keeping, that was #880