rusefi/rusefi #803: tle8888 driver bug

rusefi, 2019-05-25:
@960 reports a crash in
[image]

```
	if (isIsrContext()) {
		// this is for normal runtime
		int wasLocked = lockAnyContext();
		chSemSignalI(&tle8888_wake);
		if (!wasLocked) {
			unlockAnyContext();
		}
	} else {
		// this is for start-up to not hang up
		chSemSignal(&tle8888_wake);
	}
```

rusefi, 2019-05-25:
@dron0gus  I wonder if we do not need the whole "isIsrContext()" check, can we just always use top branch of code?

dron0gus, 2019-05-26:
@rusefi top branch was proposed by you. As a fix when this code is called from interrupt context.
Will in work for normal context?
As far as I understand from ChibiOS manuals, chSemSignal**I** will not call reschedule. It is ok for interrupt context, as re-schedule will happen on interrupt exit.
But when this code is called from normal context?

rusefi, 2019-06-08:
@960 please remind me what are steps to reproduce?

rusefi, 2019-06-09:
@mck1117 un-doing the fix since the fix breaks other things

[image]

[image]

rusefi, 2019-06-09:
this is on start up, wonder how come this worked for you with @960 while your testing?

[image]

960, 2019-06-09:
The boards I had did not run at all after some firmware changes today.
Maybe this change. 
Tried with tree different boards, with different designs.
Full crash for all, dont even boot.

960, 2019-06-09:
The tune that bricks it:

[Audi_A4_Alt.zip](https://github.com/rusefi/rusefi/files/3269099/Audi_A4_Alt.zip)

mck1117, 2019-08-24:
is this issue now fixed?

rusefi, 2019-08-24:
I assume this was resolved somewhere

closed 2019-08-24 by rusefi