rusefi/rusefi #809: software jump to DFU

rusefi, 2019-05-30:
One of the approaches to fully sealed ECU could be an ability to switch ECU into DFU mode programmatically, without physical access to BOOT* pins

We would need some way to command the firmware to go into DFU mode programmatically.

this is somewhat related to #398

rusefi, 2019-05-30:
@mck1117 @dron0gus @fpoussin any volunteers?

rusefi, 2019-06-01:
@960 says https://community.st.com/s/question/0D50X00009XkeeWSAR/stm32l476rg-jump-to-bootloader-from-software

rusefi, 2019-06-08:
see also #147

rusefi, 2019-07-28:
https://www.youtube.com/watch?v=cvKC-4tCRgw

rusefi, 2019-07-28:
https://community.st.com/s/question/0D50X00009XkezoSAB/stm32f42xx3xx-usb-device-dfu-within-a-program

rusefi, 2019-07-28:
phase 1: unconditional DFU

1) change crt0_v7m.S:

```
_crt0_entry:
// rusEfi DFU start
// Clive Two.Zero is the God of ST community forum
// Device specific, if in doubt RTFM
LDR R0, =0x40023844 // RCC_APB2ENR
LDR R1, =0x00004000 // ENABLE SYSCFG CLOCK
STR R1, [R0, #0]
LDR R0, =0x40013800 // SYSCFG_MEMRMP
LDR R1, =0x00000001 // MAP ROM AT ZERO
STR R1, [R0, #0]
LDR R0, =0x1FFF0000 // ROM BASE
LDR SP,[R0, #0] // SP @ +0
LDR R0,[R0, #4] // PC @ +4
BX R0
// rusEfi DFU end
                /* Interrupts are globally masked initially.*/
                cpsid   i
```

rusefi, 2019-07-28:
OMG magic
```// [rusefi][DFU] start
// Clive Two.Zero is the God of ST community forum
// Device specific, if in doubt RTFM
LDR R0, =0x2001FFF0 // End of SRAM for your CPU
LDR R1, =0xDEADBEEF // magic value
LDR R2, [R0, #0]
STR R0, [R0, #0] // Invalidate
CMP R2, R1
BEQ UseDFU
	bl	endOfRusEfiDFUCheck
UseDFU:
LDR R0, =0x40023844 // RCC_APB2ENR
LDR R1, =0x00004000 // ENABLE SYSCFG CLOCK
STR R1, [R0, #0]
LDR R0, =0x40013800 // SYSCFG_MEMRMP
LDR R1, =0x00000001 // MAP ROM AT ZERO
STR R1, [R0, #0]
LDR R0, =0x1FFF0000 // ROM BASE
LDR SP,[R0, #0] // SP @ +0
LDR R0,[R0, #4] // PC @ +4
BX R0
endOfRusEfiDFUCheck:
// [rusefi][DFU] end
```

rusefi, 2019-07-28:
This was tested on stm32F407

960, 2019-08-03:
I have not tried F4, but for F7 this was the working solution:

LDR R0, =0x2004FFF0 // End of SRAM for your CPU
LDR R1, =0xDEADBEEF // magic value
LDR R2, [R0, #0]
STR R0, [R0, #0] // Invalidate
CMP R2, R1
BEQ UseDFU
	// DFU bootloader not needed, jump back to normal ChibiOS startup
	bl	rusEfiAsboluteStartupMethodReturnPoint
UseDFU:
LDR     R1, =0xE000ED00
LDR     R0, =0x1FF00000
STR     R0, [R1, #8]
LDR     SP, [R0, #0]
LDR     R0, [R0, #4]
BX      R0

I also had to remove "#if EFI_SOFTWARE_DFU_JUMP" from Crt0_v7m.s to make it enter this code at all. Maybe I needed to define EFI_SOFTWARE_DFU_JUMP somewhere else than I tried.

rusefi, 2019-08-04:
@960 yes EFI_SOFTWARE_DFU_JUMP was an issue for both F4 and F7, fixed now

rusefi, 2019-08-04:
Tested on F7, seems to be working. Wow, we did it?!

closed 2019-08-04 by rusefi