rusefi/rusefi #829: does github support checkin gates?

rusefi, 2019-06-08:
Since once in a while there are pull requests which are breaking even unit tests it would be nice to have pre-merge validation of at least compilation and unit tests

rusefi, 2019-06-08:
https://devopscube.com/jenkins-build-trigger-github-pull-request/

rusefi, 2019-06-08:
Nope, I cannot figure this out :(

```
Started by user russian
Building in workspace C:\Program Files (x86)\Jenkins\workspace\pr_unit_tests
No credentials specified
 > git.exe rev-parse --is-inside-work-tree # timeout=10
Fetching changes from the remote Git repository
 > git.exe config remote.origin.url https://github.com/rusefi/rusefi # timeout=10
Fetching upstream changes from https://github.com/rusefi/rusefi
 > git.exe --version # timeout=10
 > git.exe fetch --tags --force --progress https://github.com/rusefi/rusefi +refs/pull/*:refs/remotes/origin/pr/*
 > git.exe rev-parse "origin/${sha1}^{commit}" # timeout=10
 > git.exe rev-parse "${sha1}^{commit}" # timeout=10
ERROR: Couldn't find any revision to build. Verify the repository and branch configuration for this job.
ERROR: Build step failed with exception
java.io.IOException: Cannot retrieve Git metadata for the build
	at org.jenkinsci.plugins.github.util.BuildDataHelper.getCommitSHA1(BuildDataHelper.java:87)
	at org.jenkinsci.plugins.github.status.sources.BuildDataRevisionShaSource.get(BuildDataRevisionShaSource.java:32)
	at org.jenkinsci.plugins.github.status.GitHubCommitStatusSetter.perform(GitHubCommitStatusSetter.java:135)
Caused: org.jenkinsci.plugins.github.common.CombineErrorHandler$ErrorHandlingException
	at org.jenkinsci.plugins.github.common.CombineErrorHandler.handle(CombineErrorHandler.java:74)
	at org.jenkinsci.plugins.github.status.GitHubCommitStatusSetter.perform(GitHubCommitStatusSetter.java:164)
	at hudson.tasks.BuildStepCompatibilityLayer.perform(BuildStepCompatibilityLayer.java:81)
	at hudson.tasks.BuildStepMonitor$1.perform(BuildStepMonitor.java:20)
	at hudson.model.AbstractBuild$AbstractBuildExecution.perform(AbstractBuild.java:744)
	at hudson.model.AbstractBuild$AbstractBuildExecution.performAllBuildSteps(AbstractBuild.java:690)
	at hudson.model.Build$BuildExecution.post2(Build.java:186)
	at hudson.model.AbstractBuild$AbstractBuildExecution.post(AbstractBuild.java:635)
	at hudson.model.Run.execute(Run.java:1841)
	at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
	at hudson.model.ResourceController.execute(ResourceController.java:97)
	at hudson.model.Executor.run(Executor.java:429)
Build step 'Set GitHub commit status (universal)' marked build as failure
Finished: FAILURE
```

mck1117, 2019-10-17:
Notes here about the issue, posted since we've looked at this last:

https://issues.jenkins-ci.org/browse/JENKINS-54249?page=com.atlassian.jira.plugin.system.issuetabpanels%3Aall-tabpanel

rusefi, 2019-10-17:
Sweet, good to know!

Unfortunately https://issues.jenkins-ci.org/browse/JENKINS-14530 :(

rusefi, 2019-10-17:
trying https://chuanchuanlaw.com/jenkins-how-to-manually-upgrade-jenkins-plugin-to-specific-versions/

rusefi, 2019-10-17:
GitHub Branch Source Plugin version 2.5.3
GitHub plugin version 1.29.2 is older than required. To fix, install version 1.29.3 or later.
Pipeline GitHub Notify Step Plugin version 1.0.4
GitHub Branch Source Plugin version 2.5.3 failed to load. Fix this plugin first.

rusefi, 2019-10-17:
@mck1117 fresh failure

```Running as SYSTEM
Building in workspace C:\Program Files (x86)\Jenkins\workspace\unit_tests_PRs
No credentials specified
 > git.exe rev-parse --is-inside-work-tree # timeout=10
Fetching changes from the remote Git repository
 > git.exe config remote.origin.url https://github.com/rusefi/rusefi # timeout=10
Fetching upstream changes from https://github.com/rusefi/rusefi
 > git.exe --version # timeout=10
 > git.exe fetch --tags --force --progress -- https://github.com/rusefi/rusefi +refs/pull/*:refs/remotes/origin/pr/*
ERROR: Error fetching remote repo 'origin'
hudson.plugins.git.GitException: Failed to fetch from https://github.com/rusefi/rusefi
	at hudson.plugins.git.GitSCM.fetchFrom(GitSCM.java:894)
	at hudson.plugins.git.GitSCM.retrieveChanges(GitSCM.java:1161)
	at hudson.plugins.git.GitSCM.checkout(GitSCM.java:1192)
	at hudson.scm.SCM.checkout(SCM.java:504)
	at hudson.model.AbstractProject.checkout(AbstractProject.java:1208)
	at hudson.model.AbstractBuild$AbstractBuildExecution.defaultCheckout(AbstractBuild.java:574)
	at jenkins.scm.SCMCheckoutStrategy.checkout(SCMCheckoutStrategy.java:86)
	at hudson.model.AbstractBuild$AbstractBuildExecution.run(AbstractBuild.java:499)
	at hudson.model.Run.execute(Run.java:1815)
	at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
	at hudson.model.ResourceController.execute(ResourceController.java:97)
	at hudson.model.Executor.run(Executor.java:429)
Caused by: hudson.plugins.git.GitException: Command "git.exe fetch --tags --force --progress -- https://github.com/rusefi/rusefi +refs/pull/*:refs/remotes/origin/pr/*" returned status code 1:
stdout: 
stderr: remote: Enumerating objects: 3258, done.        
remote: Counting objects:   0% (1/3258)        
remote: Counting objects:   1% (33/3258)        
remote: Counting objects:   2% (66/3258)        
remote: Counting objects:   3% (98/3258)        
remote: Counting objects:   4% (131/3258)        
remote: Counting objects:   5% (163/3258)        
remote: Counting objects:   6% (196/3258)        
remote: Counting objects:   7% (229/3258)        
remote: Counting objects:   8% (261/3258)        
remote: Counting objects:   9% (294/3258)        
remote: Counting objects:  10% (326/3258)        
remote: Counting objects:  11% (359/3258)        
remote: Counting objects:  12% (391/3258)        
remote: Counting objects:  13% (424/3258)        
remote: Counting objects:  14% (457/3258)        
remote: Counting objects:  15% (489/3258)        
remote: Counting objects:  16% (522/3258)        
remote: Counting objects:  17% (554/3258)        
remote: Counting objects:  18% (587/3258)        
remote: Counting objects:  19% (620/3258)        
remote: Counting objects:  20% (652/3258)        
remote: Counting objects:  21% (685/3258)        
remote: Counting objects:  22% (717/3258)        
remote: Counting objects:  23% (750/3258)        
remote: Counting objects:  24% (782/3258)        
remote: Counting objects:  25% (815/3258)        
remote: Counting objects:  26% (848/3258)        
remote: Counting objects:  27% (880/3258)        
remote: Counting objects:  28% (913/3258)        
remote: Counting objects:  29% (945/3258)        
remote: Counting objects:  30% (978/3258)        
remote: Counting objects:  31% (1010/3258)        
remote: Counting objects:  32% (1043/3258)        
remote: Counting objects:  33% (1076/3258)        
remote: Counting objects:  34% (1108/3258)        
remote: Counting objects:  35% (1141/3258)        
remote: Counting objects:  36% (1173/3258)        
remote: Counting objects:  37% (1206/3258)        
remote: Counting objects:  38% (1239/3258)        
remote: Counting objects:  39% (1271/3258)        
remote: Counting objects:  40% (1304/3258)        
remote: Counting objects:  41% (1336/3258)        
[... 623 lines cut ...]
	at hudson.plugins.git.GitSCM.fetchFrom(GitSCM.java:892)
	... 11 more
ERROR: Error fetching remote repo 'origin'
Discard old builds...
Sending e-mails to: edited@gmail.com
ERROR: Build step failed with exception
java.io.IOException: Cannot retrieve Git metadata for the build
	at org.jenkinsci.plugins.github.util.BuildDataHelper.getCommitSHA1(BuildDataHelper.java:34)
	at org.jenkinsci.plugins.github.status.sources.BuildDataRevisionShaSource.get(BuildDataRevisionShaSource.java:32)
	at org.jenkinsci.plugins.github.status.GitHubCommitStatusSetter.perform(GitHubCommitStatusSetter.java:135)
Caused: org.jenkinsci.plugins.github.common.CombineErrorHandler$ErrorHandlingException
	at org.jenkinsci.plugins.github.common.CombineErrorHandler.handle(CombineErrorHandler.java:74)
	at org.jenkinsci.plugins.github.status.GitHubCommitStatusSetter.perform(GitHubCommitStatusSetter.java:164)
	at hudson.tasks.BuildStepCompatibilityLayer.perform(BuildStepCompatibilityLayer.java:79)
	at hudson.tasks.BuildStepMonitor$1.perform(BuildStepMonitor.java:20)
	at hudson.model.AbstractBuild$AbstractBuildExecution.perform(AbstractBuild.java:741)
	at hudson.model.AbstractBuild$AbstractBuildExecution.performAllBuildSteps(AbstractBuild.java:690)
	at hudson.model.Build$BuildExecution.post2(Build.java:186)
	at hudson.model.AbstractBuild$AbstractBuildExecution.post(AbstractBuild.java:635)
	at hudson.model.Run.execute(Run.java:1840)
	at hudson.model.FreeStyleBuild.run(FreeStyleBuild.java:43)
	at hudson.model.ResourceController.execute(ResourceController.java:97)
	at hudson.model.Executor.run(Executor.java:429)
Build step 'Set GitHub commit status (universal)' marked build as failure
Finished: FAILURE```

rusefi, 2019-10-17:
https://dev.to/webhookrelay/automated-jenkins-builds-on-github-pull-request-4lh2 seem have done the trick!

closed 2019-10-17 by rusefi

rusefi, 2019-11-20:
4 days ago they have fixed https://issues.jenkins-ci.org/browse/JENKINS-54249

took them over a year :(