rusefi/rusefi #834: We've reinvented polymorphism in a few spots

mck1117, 2019-06-08:
There are a few spots where we have this pattern:

```
void MyType::DoThing()
{
    if(m_type == FirstType)
    {
        DoThingFirstWay();
    }
    else if (m_type = SecondType)
    {
        DoThingSecondWay();`
    }
    ... other types of thing with different behavior ...
}
```
This is why polymorphism was invented!  A vtable lookup is far faster than the above method, and is also compiler-implemented, not human implemented.  It won't miss a case, it complains if you forget to implement one, plus all modern compilers aggressively devirtualize if they can, so it will often optimize away the virtual lookup entirely.

The main example I'm thinking of is GPIO handling: `OutputPin::initPin` and `OutputPin::setValue` both use `#ifdef` and `#if` spaghetti to decide how to handle that particular pin.

In fact, the gpiochips API by @dron0gus has the `gpiochips_ops` struct, which is exactly how a vtable is formatted.

rusefi, 2019-06-09:
I have a feeling that both @dron0gus  and ChibiOS which does the same thing are simply implementing C++ manually in C for their reasons

for example
```
struct BaseSequentialStreamVMT {
  _base_sequential_stream_methods
};
```

mck1117, 2019-06-09:
That's correct, they are (which is fine). ChibiOS does it so that consumers don't have to use c++ if they don't want to, which is absolutely the right thing to do.

But rusEfi is c++, so we can use classes :)

closed 2020-12-25 by mck1117