rusefi/rusefi #849: rusEfi console sometimes crashes within SerialIoStreamJSSC

960, 2019-06-18:
JSSC java serial sometimes crashes
```
Command Line: C:\Users\Ola\Documents\RusEfi_unused\rusefi_bundle_stm32f767_osc\snapshot_20190617_2229_stm32f767_osc_rusefi\console\rusefi_console.jar

Host: Intel(R) Core(TM) i7-3632QM CPU @ 2.20GHz, 8 cores, 15G,  Windows 10 , 64 bit Build 17134 (10.0.17134.753)
Time: Tue Jun 18 04:38:31 2019 W. Europe Daylight Time elapsed time: 8 seconds (0d 0h 0m 8s)

---------------  T H R E A D  ---------------

Current thread (0x000002144eb04800):  JavaThread "communication executor1" [_thread_in_native, id=15884, stack(0x000000911af00000,0x000000911b000000)]

Stack: [0x000000911af00000,0x000000911b000000],  sp=0x000000911affe990,  free space=1018k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  com.rusefi.io.serial.SerialIoStreamJSSC.open(Ljava/lang/String;ILcom/opensr5/Logger;)Lcom/rusefi/io/serial/SerialIoStreamJSSC;+52
j  com.rusefi.io.serial.PortHolder.open(Ljava/lang/String;Lcom/opensr5/io/DataListener;)Z+7
j  com.rusefi.io.serial.PortHolder.openPort(Ljava/lang/String;Lcom/opensr5/io/DataListener;Lcom/rusefi/io/ConnectionStateListener;)Z+38
j  com.rusefi.io.serial.SerialConnector$1.run()V+20
j  java.util.concurrent.ThreadPoolExecutor.runWorker(Ljava/util/concurrent/ThreadPoolExecutor$Worker;)V+92 java.base@12.0.1
j  java.util.concurrent.ThreadPoolExecutor$Worker.run()V+5 java.base@12.0.1
j  java.lang.Thread.run()V+11 java.base@12.0.1
v  ~StubRoutines::call_stub

siginfo: EXCEPTION_ACCESS_VIOLATION (0xc0000005), reading address 0x000000004f8a3e14

```

960, 2019-06-18:
and same thing under 64 bit java
```#
# A fatal error has been detected by the Java Runtime Environment:
#
#  EXCEPTION_ACCESS_VIOLATION (0xc0000005) at pc=0x000000007110b5db, pid=9180, tid=8644
#
# JRE version: Java(TM) SE Runtime Environment (12.0.1+12) (build 12.0.1+12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (12.0.1+12, mixed mode, sharing, tiered, compressed oops, g1 gc, windows-amd64)
# Problematic frame:
# C  [jSSC-2.8_x86_64.dll+0xb5db]
#
# No core dump will be written. Minidumps are not enabled by default on client versions of Windows
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

---------------  S U M M A R Y ------------

Command Line: C:\Users\Ola\Documents\RusEfi_unused\rusefi_bundle_stm32f767_osc\snapshot_20190617_2229_stm32f767_osc_rusefi\console\rusefi_console.jar

Host: Intel(R) Core(TM) i7-3632QM CPU @ 2.20GHz, 8 cores, 15G,  Windows 10 , 64 bit Build 17134 (10.0.17134.753)
Time: Tue Jun 18 05:11:40 2019 W. Europe Daylight Time elapsed time: 8 seconds (0d 0h 0m 8s)

---------------  T H R E A D  ---------------

Current thread (0x000001d7feb0b000):  JavaThread "communication executor1" [_thread_in_native, id=8644, stack(0x000000c19bc00000,0x000000c19bd00000)]

Stack: [0x000000c19bc00000,0x000000c19bd00000],  sp=0x000000c19bcfecd0,  free space=1019k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [jSSC-2.8_x86_64.dll+0xb5db]

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  jssc.SerialNativeInterface.openPort(Ljava/lang/String;Z)J+0
j  jssc.SerialPort.openPort()Z+65
j  com.rusefi.io.serial.SerialIoStreamJSSC.open(Ljava/lang/String;ILcom/opensr5/Logger;)Lcom/rusefi/io/serial/SerialIoStreamJSSC;+52
j  com.rusefi.io.serial.PortHolder.open(Ljava/lang/String;Lcom/opensr5/io/DataListener;)Z+7
j  com.rusefi.io.serial.PortHolder.openPort(Ljava/lang/String;Lcom/opensr5/io/DataListener;Lcom/rusefi/io/ConnectionStateListener;)Z+38
j  com.rusefi.io.serial.SerialConnector$1.run()V+20
j  java.util.concurrent.ThreadPoolExecutor.runWorker(Ljava/util/concurrent/ThreadPoolExecutor$Worker;)V+92 java.base@12.0.1
j  java.util.concurrent.ThreadPoolExecutor$Worker.run()V+5 java.base@12.0.1
j  java.lang.Thread.run()V+11 java.base@12.0.1
v  ~StubRoutines::call_stub

siginfo: EXCEPTION_ACCESS_VIOLATION (0xc0000005), reading address 0xffffffffff5641d7

Register to memory mapping:

RIP=0x000000007110b5db jSSC-2.8_x86_64.dll
RAX=0xffffffffff5641d7 is an unknown value
RBX=0x000000c19bcfed40 is pointing into the stack for thread: 0x000001d7feb0b000
RCX=0x00007ffeea51ab84 ntdll.dll
RDX=0x0 is NULL
RSP=0x000000c19bcfecd0 is pointing into the stack for thread: 0x000001d7feb0b000
RBP=0x000000c19bcfee50 is pointing into the stack for thread: 0x000001d7feb0b000
RSI=0x000001d7feb0b330 points into unknown readable memory: e0 1d f9 8f fe 7f 00 00
RDI=0x0000000000000d44 is an unknown value
R8 =0x000000c19bcfebc8 is pointing into the stack for thread: 0x000001d7feb0b000
R9 =0x000000c19bcfecc0 is pointing into the stack for thread: 0x000001d7feb0b000
```

960, 2019-06-18:
Crash is pretty reproducible with SerialIoStreamJSSC

On the other hand SerialIoStreamJSerialComm seems to be working fine!

rusefi, 2019-06-18:
Wow, suddenly SerialIoStreamJSerialComm is useful!

I will have to make some way to run console with SerialIoStreamJSerialComm instead of SerialIoStreamJSSC

rusefi, 2019-06-22:
I have same on my fresh Windows 10 64 bit

rusefi, 2019-06-22:
@andreika-git wonders "Is there any reason not to switch to the new connector and forget about the old one?"

A: it looks like new connector has way more logic than new connector.
Open task is to do the gap analysis on what code is in the old connector which should be extracted or removed

rusefi, 2019-06-23:
See also #858

closed 2019-06-23 by rusefi