rusefi/rusefi #852: setDefaultBoardConfiguration shouldn't have Frankenso pinout

mck1117, 2019-06-21:
Right now the configuration reset (`resetConfigurationExt`) does the following:
1) `setDefaultConfiguration`: This sets up normal defaults for most engine parameters, but not hardware.
2) If the engine type is NOT set to one of a short list, call `setDefaultBoardConfiguration`.  This sets up what appears to be hardware configuration (including pinout) for Frankenso.
3) `setBoardConfigurationOverrides`: This then _replaces_ any settings set by 2) with hardware settings specific to this board.
4) apply engine-specific defaults.  The default engine, `CUSTOM_ENGINE`, sets even more Frankenso-specific settings.  There's even a comment here noting that this is Frankenso-specific stuff.  Some non-custom engines also set pinouts and other hardware configuration.

There are a few problems here:
- If a board's `setBoardConfigurationOverrides` in step 3 forgets something, you might have a Frankenso default connected to the wrong hardware, dangerous hardware, or nothing at all.
- Any boards that don't want to use a feature have to un-set it in step 3.
- If engine-specific defaults are supposed to work, then they can't set any hardware configuration like pinouts.

There are incremental changes to make here to improve this, roughly in this order:
1) Make the existing `setDefaultBoardConfiguration` belong to only Frankenso.  Rename `setXXXXXOverrides` to something like `setXXXXXConfiguration`.  Omit any un-setting of features on a per-board basis, since they'll no longer be set in the first place.  (this is pretty safe - unnecessary clearing of settings won't cause problems, and if a board really needs to build off of the frankenso config, it can call that first)
2) Figure out what to do about engine-specific defaults.  Maybe that means remove any hardware config from them.

960, 2019-06-22:
The safest option are to keep all the engines stuff away from other boards completly.

I still think this is a dangerous evil thing to have in firmware at all.

Imagine you actually use one of the presets, but have the trigger/timing at different angle.

Then a firmware reset may not be noticed at all..

Before the engine blows!

rusefi, 2019-07-25:
@mck1117 thank you for the detailed problem statement it definitely made improving things easy

what do you think about current state of master?

rusefi, 2019-09-10:
@mck1117 ?

closed 2019-09-10 by mck1117