rusefi/rusefi #854: ADC issue on F7

rusefi, 2019-06-22:
@mck1117 believes that the ADC issues @960 has originally reported are related to F7 memory caching mechanism

rusefi, 2019-06-22:
@mck1117 @960 @andreika-git @dron0gus 

Here is ChibiOS code
```
/* TRUE means that DMA-accessible buffers are placed in a non-cached RAM
   area and that no cache management is required.*/
#define DMA_BUFFERS_COHERENCE TRUE

#if !DMA_BUFFERS_COHERENCE
/* Note, the buffer is aligned to a 32 bytes boundary because limitations
   imposed by the data cache. Note, this is GNU specific, it must be
   handled differently for other compilers.
   Only required if the ADC buffer is placed in a cache-able area.*/
#if defined(__GNUC__)
__attribute__((aligned (32)))
#endif
#endif
static adcsample_t samples1[ADC_GRP1_NUM_CHANNELS * ADC_GRP1_BUF_DEPTH];

size_t nx = 0, ny = 0;
static void adccallback(ADCDriver *adcp, adcsample_t *buffer, size_t n) {

#if !DMA_BUFFERS_COHERENCE
  /* DMA buffer invalidation because data cache, only invalidating the
     half buffer just filled.
     Only required if the ADC buffer is placed in a cache-able area.*/
  dmaBufferInvalidate(buffer,
                      n * adcp->grpp->num_channels * sizeof (adcsample_t));
#else
  (void)adcp;
#endif
```

I believe it would make more sense for us to flip the meaning of this variable and have it like

#define EFI_NEED_DMA_BUFFERS_PROTECTION TRUE

Opinions?

960, 2019-06-22:
I tested with 

#define DMA_BUFFERS_COHERENCE TRUE

today.

Nothing at all on the board works, and I get Fatal error after about 20 seconds..

PR #855 (2019-06-23): Fix F7 ADC #854
On ARM Cortex M7, only the CPU can invalidate the CPU cache.  So when DMA writes to SRAM, the CPU doesn't know it should re-read.  This change adds an invalidate to the ADC's sample buffer before reading it, so we force a read from SRAM, actually loading the new values.

Fixes #854
closed 2019-06-23 by rusefi