rusefi/rusefi #867: Refactoring: OS assess should be explicit

rusefi, 2019-07-04:
At the moment all source code files get access to ChibiOS via global.h

In order to separate synchronous from asynchronous it would be nice to explicitly state OS header access

mck1117, 2019-07-05:
It would also be great if we could avoid `#include "ch.h"` in every file.  That's most of what's killing the compile time.

mck1117, 2019-07-06:
> In order to separate synchronous from asynchronous

I'm not sure what you mean here

rusefi, 2019-07-06:
@mck1117 just expanded the description a bit

mck1117, 2019-07-06:
BTW this sort of thing was the intention of #645.  We should discuss elsewhere, but there are far more C++ friendly ways to do this than #ifdef macro spaghetti.

rusefi, 2019-07-06:
I am stuck, things are not going well :(
On the one hand engine.h depends on event_registry.h which depends on OutputPin which has "os / hal" variable ioportid_t port;

without hal.h  ioportid_t would not compile. I am NOT excited about doing
```
#ifndef ioportid_t 
typedef ioportid_t int32
#endif
```

I cannot simply include #include "hal_pal.h" since PAL depends on OS heavily and for #include "hal_pal.h" to compile i seem to need complete ch.h

There is probably a way if engine.h would use pointers to OutputPin not instance and if I use shadow unknown type for pointers but that's a bit nasty.

Overall, do not have a trivial implementation :(