rusefi/rusefi #869: CAN not working with new MRE setup

960, 2019-07-05:
Not sure where the issue are(HW/SW), but I cant get CAN working now.

I have tried all possible settings, can sniffer and logic analyzer.

With can sniffer, there are nothing.

With logic analyzer I see traffic at CAN L, but nothing at CAN H.

mck1117, 2019-07-05:
What does the signal look like on CANL?

960, 2019-07-05:
It's a logic signal that looks rational.

Just looked thru mcuconf.h for F767, and it does not look like we have can enabled.
Tried to add this, but still not working.

#define STM32_CAN_USE_CAN1                  TRUE
#define STM32_CAN_USE_CAN2                  TRUE

rusefi, 2019-07-05:
@960 something seems wrong. Please paste exact github URL which you believe does not have CAN.

960, 2019-07-05:
I had looked in wrong file.

So, I am back to having no idea at all what could be wrong

960, 2019-07-06:
Can this be related to F7 having different speed, and wrong prescaler?

rusefi, 2019-07-06:
@960 I think yes, it could be related to speed and prescaler

mck1117, 2019-07-06:
I've confirmed with scope that F7 CAN is running 29% too fast, 1.29 = 216mhz/168mhz

960, 2019-07-06:
When I connect a instrument cluster at bench now, the read counter in TS starts counting.

No output at all still.
Nothing at can-sniffer.

The send counter in TS goes to 3 at startup, and thats it.

mck1117, 2019-07-06:
@rusefi I'm not sure this is closed, he hand applied #872 and that didn't seem to fix it

960, 2019-07-06:
Still not working.

Do you capture anything at the MRE output?

960, 2019-07-06:
Now it's finally solved.

It was actually a bit more complicated..

rusefi, 2019-07-06:
@960 so what was it?

rusefi, 2019-07-07:
https://rusefi.slack.com/messages/C4P8TSCJF/details/ says we have no idea what was wrong (if anything?) on top of scaling

PR #872 (2019-07-06): Add per-MCU (f4 vs f7) CAN settings
This fixes #869.
closed 2019-07-07 by rusefi