rusefi/rusefi #883: VVT support for VAG trigger

rusefi, 2019-07-12:
@960 is asking for a new VVT trigger. Tooth shape is available at https://rusefi.com/forum/viewtopic.php?f=5&t=1602

VVT trigger system overall could use some refactoring to reduce code duplication with normal trigger

960, 2019-07-25:
Are there any progress on this?

rusefi, 2019-11-29:
http://www.maxxecu.com/webhelp/trigger_system-vw_18t_cam.html

mck1117, 2020-08-24:
I think this is the same as #1368

closed 2020-08-27 by rusefillc

960, 2020-12-09:
I still havent got this working..

Tried every setting with inverted etc..

Usually it only shows vvt position of 0, but sometimes it shows a large value of 9xx.
Only a static value.

I am just controlling VVT to two positions (on/off) based on rpm, but would be cool to have it working

rusefillc, 2020-12-09:
hard to test without tune on rusEFI online, could also he a HW issue

960, 2020-12-09:
Hw input looks good in both Logic analyzer and Logger:

[image]

rusefillc, 2020-12-09:
VVT tuning usability is a mess, hard to tell without your tune on rusEFI online

commit 327b67c1 (2020-08-27): VVT support for VAG trigger fix #883