rusefi/rusefi #884: FATAL error: lsize mismatch 336/3342

rusefi, 2019-07-12:
got this from integration testing run today:

```   [java] Fri Jul 12 08:41:11 EDT 2019<EOT>: text body: waiting for 59 byte(s)
     [java] Fri Jul 12 08:41:11 EDT 2019<EOT>: msg,FATAL error: lsize mismatch 336/3342 [0.00/0.00,],
     [java] Fri Jul 12 08:41:11 EDT 2019<EOT>: postMessage EngineState: FATAL error: lsize mismatch 336/3342 [0.00/0.00
     [java] Fri Jul 12 08:41:11 EDT 2019<EOT>: EngineState.unknown: ],
     [java] Fri Jul 12 08:41:11 EDT 2019<EOT>: FAILURE: Not true: Fatal not expected
     [java] java.lang.IllegalStateException: Not true: Fatal not expected
     [java]     at com.rusefi.TestingUtils.fail(TestingUtils.java:52)
     [java]     at com.rusefi.TestingUtils.assertTrue(TestingUtils.java:27)
     [java]     at com.rusefi.TestingUtils.assertNull(TestingUtils.java:104)
     [java]     at com.rusefi.AutoTest.sendCommand(AutoTest.java:435)
     [java]     at com.rusefi.AutoTest.sendCommand(AutoTest.java:431)
     [java]     at com.rusefi.AutoTest.setEngineType(AutoTest.java:131)
     [java]     at com.rusefi.AutoTest.testFordAspire(AutoTest.java:308)
     [java]     at com.rusefi.AutoTest.mainTestBody(AutoTest.java:43)
     [java]     at com.rusefi.RealHwTest.runRealHardwareTest(RealHwTest.java:62)
     [java]     at com.rusefi.RealHwTest.main(RealHwTest.java:26)```

is this some threading issue in loggingcentral? :(

rusefi, 2019-08-06:
``` [java] 2019-08-06_17_54_27_940<EOT>: IncomingDataBuffer: 107 byte(s) arrived
     [java] 2019-08-06_17_54_27_940<EOT>: Sending packet 47 
     [java] 2019-08-06_17_54_27_940<EOT>: text header: waiting for 2 byte(s)
     [java] 2019-08-06_17_54_27_940<EOT>: IncomingDataBuffer: 60 byte(s) arrived
     [java] 2019-08-06_17_54_27_940<EOT>: text body: waiting for 58 byte(s)
     [java] 2019-08-06_17_54_27_940<EOT>: msg,FATAL error: lsize mismatch 74/3668 [order 0/1,],
     [java] 2019-08-06_17_54_27_940<EOT>: postMessage EngineState: FATAL error: lsize mismatch 74/3668 [order 0/1
     [java] 2019-08-06_17_54_27_940<EOT>: EngineState.unknown: ],
     [java] 2019-08-06_17_54_28_050<EOT>: Sending packet 4F 00 00 64 01 
     [java] 2019-08-06_17_54_28_050<EOT>: output channels header: waiting for 2 byte(s)
     [java] 2019-08-06_17_54_28_050<EOT>: IncomingDataBuffer: 256 byte(s) arrived
     [java] 2019-08-06_17_54_28_050<EOT>: output channels body: waiting for 361 byte(s)
     [java] 2019-08-06_17_54_28_050<EOT>: IncomingDataBuffer: 107 byte(s) arrived
     [java] 2019-08-06_17_54_28_050<EOT>: Sending packet 47 
     [java] 2019-08-06_17_54_28_050<EOT>: text header: waiting for 2 byte(s)
     [java] 2019-08-06_17_54_28_050<EOT>: IncomingDataBuffer: 7 byte(s) arrived```

rusefi, 2019-11-30:
Just happened again :( ``` [java] 2019-11-29_20_28_10_412<EOT>: postMessage EngineState: FATAL error: lsize mismatch 109/3816 [order 0/1
     [java] 2019-11-29_20_28_10_412<EOT>: EngineState.unknown: ],```

rusefi, 2019-12-29:
Still an issue obviously

``` [java] 2019-12-29_13_27_36_373<EOT>: text body: waiting for 59 byte(s)
     [java] 2019-12-29_13_27_36_373<EOT>: msg,FATAL error: lsize mismatch 287/3995 [0.00/0.00,],
     [java] 2019-12-29_13_27_36_373<EOT>: postMessage EngineState: FATAL error: lsize mismatch 287/3995 [0.00/0.00
     [java] 2019-12-29_13_27_36_373<EOT>: EngineState.unknown: ],
     [java] 2019-12-29_13_27_36_464<EOT>: Sending packet 4F 00 00 DC 00
     [java] 2019-12-29_13_27_36_464<EOT>: output channels header: waiting for 2 byte(s)
     [java] 2019-12-29_13_27_36_465<EOT>: IncomingDataBuffer: 227 byte(s) arrived
     [java] 2019-12-29_13_27_36_466<EOT>: output channels body: waiting for 225 byte(s)
     [java] 2019-12-29_13_27_36_466<EOT>: Sending packet 47```

rusefillc, 2020-10-17:
has been a while :)

```
     [java] I 201016 232015.806 [communication executor1] IoStream - [serial] output channels body: waiting for 293 byte(s)
     [java] I 201016 232015.807 [communication executor1] IoStream - text header: waiting for 2 byte(s)
     [java] I 201016 232015.807 [Serial Port Event Thread] IoStream - IncomingDataBuffer: 81 byte(s) arrived
     [java] I 201016 232015.807 [communication executor1] IoStream - [serial] text body: waiting for 79 byte(s)
     [java] I 201016 232015.807 [communication executor1] LinkManager - msg,CRITICAL error: lsize mismatch 1036/3251 [0.00/0.00,] 20201017@source,
     [java] I 201016 232015.807 [communication executor1] MessagesCentral - postMessage EngineState: CRITICAL error: lsize mismatch 1036/3251 [0.00/0.00
     [java] I 201016 232015.807 [communication executor1] EngineState - EngineState.unknown: ] 20201017@source,
     [java] I 201016 232015.903 [main] TestingUtils - FAILURE: Not true: Fatal not expected
     [java] I 201016 232015.903 [communication executor1] IoStream - output channels header: waiting for 2 byte(s)
```

closed 2021-04-18 by rusefillc

mck1117, 2022-01-18:
somebody saw it again today

[image]

PR #2439 (2021-04-18): rewrite text logging system
No longer require a logging buffer for every thread that needs one.  Instead, use a queue of buffers so one thread can't block execution (particularly of interrupts!) when it writes to the log.

Uses ChibiOS message queues to maintain one queue of free buffers, and one queue of used buffers.  When a thread wants to write, it acquires a free buffer, prints to it, and pushes it in to the used queue.  A dedicated thread then dequeues and writes lines from the used buffer in to the large output buffer.

probably fixes #884 too

fixes #2524