rusefi/rusefi #892: MRE default pinout

Abricosvw, 2019-07-28:
we need default MRE pinout in default MRE bundle

Abricosvw, 2019-07-28:
current MRE#2 board has
```2019-07-28_14_25_15_124: EngineState: confirmation_pins:4
2019-07-28_14_25_15_124: EngineState: pin PA5: trigger#1
2019-07-28_14_25_15_125: EngineState: pin PA10: USB ID
2019-07-28_14_25_15_126: EngineState: pin PA11: USB DM
2019-07-28_14_25_15_128: EngineState: pin PA12: USB DP
2019-07-28_14_25_15_130: EngineState: pin PB3: SPI clock
2019-07-28_14_25_15_131: EngineState: pin PB4: SPI master in 
2019-07-28_14_25_15_132: EngineState: pin PB5: SPI master out
2019-07-28_14_25_15_132: EngineState: pin PB14: led: ERROR status
2019-07-28_14_25_15_132: EngineState: pin PC1: adc input
2019-07-28_14_25_15_132: EngineState: pin PC2: adc input
2019-07-28_14_25_15_132: EngineState: pin PC6: trigger#2
2019-07-28_14_25_15_133: EngineState: pin PD1: coil4
2019-07-28_14_25_15_133: EngineState: pin PD2: coil3
2019-07-28_14_25_15_133: EngineState: pin PD5: tle8888 CS
2019-07-28_14_25_15_135: EngineState: pin PD6: coil2
2019-07-28_14_25_15_137: EngineState: pin PD7: coil1
2019-07-28_14_25_15_142: EngineState: pin PE1: trg_err
2019-07-28_14_25_15_142: EngineState: pin PE2: led: comm status
2019-07-28_14_25_15_142: EngineState: pin PE4: led: running status
2019-07-28_14_25_15_142: EngineState: pin PE11: injector4
2019-07-28_14_25_15_142: EngineState: pin PE12: injector3
2019-07-28_14_25_15_143: EngineState: pin PE13: injector2
2019-07-28_14_25_15_143: EngineState: pin PE14: injector1
2019-07-28_14_25_15_143: EngineState: ext (null).-1: free
2019-07-28_14_25_15_143: EngineState: ext (null).-1: free
2019-07-28_14_25_15_144: EngineState: ext (null).-1: free
2019-07-28_14_25_15_144: EngineState: ext (null).-1: free
2019-07-28_14_25_15_144: EngineState: ext (null).-1: free
2019-07-28_14_25_15_144: EngineState: ext (null).-1: free
2019-07-28_14_25_15_144: EngineState: ext (null).-1: free
2019-07-28_14_25_15_145: EngineState: ext (null).-1: free
2019-07-28_14_25_15_145: EngineState: ext (null).-1: free
2019-07-28_14_25_15_145: EngineState: ext (null).-1: free
2019-07-28_14_25_15_145: EngineState: ext (null).-1: free
2019-07-28_14_25_15_145: EngineState: ext (null).-1: free
2019-07-28_14_25_15_146: EngineState: ext (null).-1: free
2019-07-28_14_25_15_148: EngineState: ext (null).-1: free
2019-07-28_14_25_15_152: EngineState: ext (null).-1: free
2019-07-28_14_25_15_152: EngineState: ext (null).-1: free
2019-07-28_14_25_15_152: EngineState: ext (null).-1: free
2019-07-28_14_25_15_152: EngineState: ext (null).-1: free
2019-07-28_14_25_15_153: EngineState: ext (null).-1: free
2019-07-28_14_25_15_153: EngineState: ext (null).-1: free
2019-07-28_14_25_15_153: EngineState: ext (null).-1: free
2019-07-28_14_25_15_153: EngineState: ext (null).-1: free
2019-07-28_14_25_15_153: EngineState: ext (null).-1: free
2019-07-28_14_25_15_154: EngineState: ext (null).-1: free
2019-07-28_14_25_15_154: EngineState: Total pins count: 23
2019-07-28_14_25_16_724: EngineState: WARNING: MAF sensor needed for current fuel algorithm
```

mck1117, 2019-08-04:
I believe this is now done - try again with a newer microRusefi build of the firmware.  If I missed something in my PR #895, please let me know!

closed 2019-09-10 by rusefi