rusefi/rusefi #897: Internal trigger simulator shouldn't be stored in persistent config

mck1117, 2019-08-03:
Suppose an unsuspecting user (in this case, me) does the following:

1) Plug ECU in on the bench, activate self stimulation to investigate why the injectors aren't firing.
2) Finds & fixes the problem.
3) Plug ECU in to car, turn on ignition. Injectors and coils start firing, what's going on?

I propose that we move "enable self stimulation" to be a button on the "Bench Test" page, instead of a persistent setting that's flashed to the controller.

rusefi, 2019-08-03:
board testing firmware is a special case but up for debate if even it should be persisted. maybe there should be two ways - more visible "enable temporary self stimulation" and more hidden persisted self stimulation

mck1117, 2019-12-11:
I did it again yesterday :grin:

I plugged the ECU in to the car, turned the key to on, and heard a lot of injector clicking...

mck1117, 2020-04-10:
+1 for users who've been tripped up by this...

rusefi, 2020-04-10:
Jimmy lunztel was confused by a board in QA configuration

mck1117, 2020-04-18:
+1 for this happening to a user.  Happened again.

mck1117, 2020-04-24:
And one more.

PR #1356 (2020-04-24): don't store self stim in config
Fixes #897.  Now a button instead of a setting.
closed 2020-04-24 by rusefi