rusefi/rusefi #904: Reinitialize TLE8888 if it resets

mck1117, 2019-08-24:
Under some conditions (like the ISO 16750-2 cranking tests), the supply voltage on microRusefi can get low enough to trip internal reset on the tle8888.  Should this happen, we lose the ability to drive injectors/etc.  The voltage stays high enough to supply the stm32, but the blip is too short to measure with our current analog input circuitry.

Before implementing something naïve, we should come up with some sensible logic that doesn't rely on a fixed voltage setting, or the ability to detect a very short pulse.

mck1117, 2019-08-24:
FYI - this isn't a hardware defect necessarily, just a limitation of the current design.  Without a boost regulator, there isn't very much we can do during a >1ms voltage dip below 5v. The observed dip gets down to less than 4 volts for up to 20ms, consistent with the ISO tests.

rusefi, 2019-09-05:
Looks like what I have is in fact a reset condition, here are the facts:

1) fuel pump always turns on when I turn ignition on - so, TLE8888 always _starts_
2) but often car would not start with cranking and fuel pump bench testing fails - so, cranking affects TLE8888 functionality negatively.

closed 2019-11-13 by mck1117