rusefi/rusefi #917: TLE8888 watchdog

rusefi, 2019-09-05:
watchdog is a nice safety feature, we shall implement the code to support relevant version of TLE8888 chip

shadowm60, 2019-12-05:
Is the window watchdog the target here? Where should we trigger, the basic ECU sw could function even without the OS, so that means that we could trigger WD from something cyclic like a task, but it is really relevant?

rusefi, 2020-02-10:
mouser might be telling us that eventually WD would be required since only the PN with WD would be available?

[image]

rusefi, 2020-02-10:
[image]

rusefi, 2020-02-11:
[image]

rusefi, 2020-02-11:
[image]

rusefi, 2020-02-11:
2QK chip also responds

[image]

rusefi, 2020-02-11:
new debug command:

2020-02-11_17_55_19_057: EngineState: confirmation_tle8888:7
2020-02-11_17_55_19_058: EngineState: 00: 00
2020-02-11_17_55_19_059: EngineState: 00: 00
2020-02-11_17_55_19_059: EngineState: 02: 00
2020-02-11_17_55_19_060: EngineState: 04: 00
2020-02-11_17_55_19_060: EngineState: 06: 00
2020-02-11_17_55_19_060: EngineState: 08: 00
2020-02-11_17_55_19_061: EngineState: 0A: 00
2020-02-11_17_55_19_061: EngineState: 0C: 00
2020-02-11_17_55_19_062: EngineState: 0E: 00
2020-02-11_17_55_19_063: EngineState: 10: 00
2020-02-11_17_55_19_063: EngineState: 12: 00
2020-02-11_17_55_19_064: EngineState: 14: 00
2020-02-11_17_55_19_064: EngineState: 16: 00
2020-02-11_17_55_19_065: EngineState: 18: 00
2020-02-11_17_55_19_065: EngineState: 1A: 00
2020-02-11_17_55_19_066: EngineState: 1C: 00
2020-02-11_17_55_19_066: EngineState: 1E: 00
2020-02-11_17_55_19_067: EngineState: 20: 00
2020-02-11_17_55_19_068: EngineState: 22: 00
2020-02-11_17_55_19_068: EngineState: 24: 00
2020-02-11_17_55_19_068: EngineState: 26: 00
2020-02-11_17_55_19_069: EngineState: 28: 00
2020-02-11_17_55_19_069: EngineState: 2A: 00
2020-02-11_17_55_19_069: EngineState: 2C: 00
2020-02-11_17_55_19_070: EngineState: 2E: 00
2020-02-11_17_55_19_070: EngineState: 30: 00
2020-02-11_17_55_19_070: EngineState: 32: 00
2020-02-11_17_55_19_071: EngineState: 34: 00
2020-02-11_17_55_19_071: EngineState: 36: 00
2020-02-11_17_55_19_071: EngineState: 38: 00
2020-02-11_17_55_19_072: EngineState: 3A: 00
2020-02-11_17_55_19_072: EngineState: 3C: 00
2020-02-11_17_55_19_072: EngineState: 3E: 00
2020-02-11_17_55_19_073: EngineState: 40: 02
2020-02-11_17_55_19_073: EngineState: 42: 00
2020-02-11_17_55_19_073: EngineState: 44: 00
2020-02-11_17_55_19_074: EngineState: 46: 00
2020-02-11_17_55_19_074: EngineState: 48: 00
2020-02-11_17_55_19_079: EngineState: 4A: 00
2020-02-11_17_55_19_080: EngineState: 4C: AA
2020-02-11_17_55_19_113: EngineState: 4E: AA
2020-02-11_17_55_19_115: EngineState: 50: AA
2020-02-11_17_55_19_116: EngineState: 52: AA
2020-02-11_17_55_19_121: EngineState: 54: 00
2020-02-11_17_55_19_128: EngineState: 56: 82
2020-02-11_17_55_19_128: EngineState: 58: 00
2020-02-11_17_55_19_132: EngineState: 5A: 00
2020-02-11_17_55_19_134: EngineState: 5C: 00
2020-02-11_17_55_19_135: EngineState: 5E: 00
2020-02-11_17_55_19_141: EngineState: 60: 00
2020-02-11_17_55_19_145: EngineState: 62: 00
2020-02-11_17_55_19_145: EngineState: 64: 00
2020-02-11_17_55_19_150: EngineState: 66: 00
2020-02-11_17_55_19_172: EngineState: 68: 19
2020-02-11_17_55_19_172: EngineState: 6A: 60
2020-02-11_17_55_19_174: EngineState: 6C: 00
2020-02-11_17_55_19_177: EngineState: 6E: 00
2020-02-11_17_55_19_177: EngineState: 70: 1F
2020-02-11_17_55_19_180: EngineState: 72: 00
2020-02-11_17_55_19_206: EngineState: 74: 80
2020-02-11_17_55_19_206: EngineState: 6A: 60
2020-02-11_17_55_19_209: EngineState: 78: 00
2020-02-11_17_55_19_213: EngineState: 7A: 00
2020-02-11_17_55_19_213: EngineState: 7C: 00
2020-02-11_17_55_19_216: EngineState: 7E: 00
2020-02-11_17_55_19_219: EngineState: 00: FF
2020-02-11_17_55_19_220: EngineState: 02: 3F
2020-02-11_17_55_19_223: EngineState: 04: 3F
2020-02-11_17_55_19_225: EngineState: 06: 30
2020-02-11_17_55_19_225: EngineState: 08: 3F
2020-02-11_17_55_19_229: EngineState: 0A: 3F
2020-02-11_17_55_19_232: EngineState: 0C: 00
2020-02-11_17_55_19_233: EngineState: 0E: 00
2020-02-11_17_55_19_241: EngineState: 10: 00
2020-02-11_17_55_19_245: EngineState: 12: 00
2020-02-11_17_55_19_247: EngineState: 14: 00
2020-02-11_17_55_19_247: EngineState: 16: 00
2020-02-11_17_55_19_252: EngineState: 18: 00
2020-02-11_17_55_19_254: EngineState: 1A: 00
2020-02-11_17_55_19_255: EngineState: 1C: 09
2020-02-11_17_55_19_256: EngineState: 1E: A4
2020-02-11_17_55_19_256: EngineState: 20: 0D
2020-02-11_17_55_19_256: EngineState: 22: 00
2020-02-11_17_55_19_257: EngineState: 24: 00
2020-02-11_17_55_19_257: EngineState: 26: 00
2020-02-11_17_55_19_258: EngineState: 28: 01
2020-02-11_17_55_19_270: EngineState: 2A: 10
2020-02-11_17_55_19_270: EngineState: 2C: 0A
2020-02-11_17_55_19_273: EngineState: 2E: 3F
2020-02-11_17_55_19_275: EngineState: 30: 40
2020-02-11_17_55_19_276: EngineState: 32: 10
2020-02-11_17_55_19_279: EngineState: 34: 0F
2020-02-11_17_55_19_282: EngineState: 36: 00
2020-02-11_17_55_19_282: EngineState: 38: 00
2020-02-11_17_55_19_284: EngineState: 3A: 00
2020-02-11_17_55_19_287: EngineState: 3C: 00
2020-02-11_17_55_19_287: EngineState: 3E: FF
2020-02-11_17_55_19_288: EngineState: 40: 77
2020-02-11_17_55_19_289: EngineState: 42: F7
2020-02-11_17_55_19_291: EngineState: 44: 77
2020-02-11_17_55_19_296: EngineState: 46: 47
2020-02-11_17_55_19_299: EngineState: 48: 1B
2020-02-11_17_55_19_300: EngineState: 4A: 00
2020-02-11_17_55_19_302: EngineState: 4C: 00
2020-02-11_17_55_19_305: EngineState: 4E: 00
2020-02-11_17_55_19_305: EngineState: 50: 00
2020-02-11_17_55_19_308: EngineState: 52: 00
2020-02-11_17_55_19_310: EngineState: 54: 00
2020-02-11_17_55_19_311: EngineState: 56: 00
2020-02-11_17_55_19_313: EngineState: 58: 00
2020-02-11_17_55_19_316: EngineState: 5A: 00
2020-02-11_17_55_19_318: EngineState: 5C: 00
2020-02-11_17_55_19_319: EngineState: 5E: 00
2020-02-11_17_55_19_322: EngineState: 60: 00
2020-02-11_17_55_19_324: EngineState: 62: 00
2020-02-11_17_55_19_325: EngineState: 64: 00
2020-02-11_17_55_19_328: EngineState: 66: 00
2020-02-11_17_55_19_331: EngineState: 68: 00
2020-02-11_17_55_19_332: EngineState: 6A: 00
2020-02-11_17_55_19_335: EngineState: 6C: 00
2020-02-11_17_55_19_337: EngineState: 6E: 00
2020-02-11_17_55_19_338: EngineState: 70: 00
2020-02-11_17_55_19_341: EngineState: 72: 00
2020-02-11_17_55_19_341: EngineState: 74: 00
2020-02-11_17_55_19_348: EngineState: 76: 00
2020-02-11_17_55_19_351: EngineState: 78: 00
2020-02-11_17_55_19_354: EngineState: 7A: 40

rusefi, 2020-02-11:
0.4.5 board -2QK chip

2020-02-11_18_16_16_084: EngineState: confirmation_tle8888:7
2020-02-11_18_16_16_087: EngineState: 00: 00
2020-02-11_18_16_16_088: EngineState: 00: 00
2020-02-11_18_16_16_088: EngineState: 02: 00
2020-02-11_18_16_16_089: EngineState: 04: 00
2020-02-11_18_16_16_090: EngineState: 06: 00
2020-02-11_18_16_16_090: EngineState: 08: 00
2020-02-11_18_16_16_091: EngineState: 0A: 00
2020-02-11_18_16_16_091: EngineState: 0C: 00
2020-02-11_18_16_16_091: EngineState: 0E: 00
2020-02-11_18_16_16_092: EngineState: 10: 00
2020-02-11_18_16_16_092: EngineState: 12: 00
2020-02-11_18_16_16_093: EngineState: 14: 00
2020-02-11_18_16_16_093: EngineState: 16: 00
2020-02-11_18_16_16_093: EngineState: 18: 00
2020-02-11_18_16_16_094: EngineState: 1A: 00
2020-02-11_18_16_16_094: EngineState: 1C: 00
2020-02-11_18_16_16_095: EngineState: 1E: 00
2020-02-11_18_16_16_095: EngineState: 20: 00
2020-02-11_18_16_16_095: EngineState: 22: 00
2020-02-11_18_16_16_096: EngineState: 24: 00
2020-02-11_18_16_16_097: EngineState: 26: 00
2020-02-11_18_16_16_098: EngineState: 28: 00
2020-02-11_18_16_16_098: EngineState: 2A: 00
2020-02-11_18_16_16_099: EngineState: 2C: 00
2020-02-11_18_16_16_099: EngineState: 2E: 00
2020-02-11_18_16_16_099: EngineState: 30: 00
2020-02-11_18_16_16_100: EngineState: 32: 00
2020-02-11_18_16_16_100: EngineState: 34: 00
2020-02-11_18_16_16_101: EngineState: 36: 00
2020-02-11_18_16_16_101: EngineState: 38: 00
2020-02-11_18_16_16_101: EngineState: 3A: 00
2020-02-11_18_16_16_102: EngineState: 3C: 00
2020-02-11_18_16_16_102: EngineState: 3E: 00
2020-02-11_18_16_16_103: EngineState: 40: 02
2020-02-11_18_16_16_103: EngineState: 42: 00
2020-02-11_18_16_16_103: EngineState: 44: 00
2020-02-11_18_16_16_104: EngineState: 46: 00
2020-02-11_18_16_16_104: EngineState: 48: 00
2020-02-11_18_16_16_104: EngineState: 4A: 00
2020-02-11_18_16_16_105: EngineState: 4C: AA
2020-02-11_18_16_16_111: EngineState: 4E: AA
2020-02-11_18_16_16_111: EngineState: 50: AA
2020-02-11_18_16_16_112: EngineState: 52: AA
2020-02-11_18_16_16_112: EngineState: 54: 00
2020-02-11_18_16_16_113: EngineState: 56: 82
2020-02-11_18_16_16_113: EngineState: 58: 00
2020-02-11_18_16_16_113: EngineState: 5A: 00
2020-02-11_18_16_16_114: EngineState: 5C: 00
2020-02-11_18_16_16_122: EngineState: 5E: 00
2020-02-11_18_16_16_122: EngineState: 60: 00
2020-02-11_18_16_16_123: EngineState: 6A: 60
2020-02-11_18_16_16_123: EngineState: 6A: 60
2020-02-11_18_16_16_124: EngineState: 66: 00
2020-02-11_18_16_16_124: EngineState: 68: 19
2020-02-11_18_16_16_125: EngineState: 6A: 60
2020-02-11_18_16_16_125: EngineState: 6C: 30
2020-02-11_18_16_16_126: EngineState: 6E: 30
2020-02-11_18_16_16_126: EngineState: 70: 30
2020-02-11_18_16_16_127: EngineState: 72: 30
2020-02-11_18_16_16_127: EngineState: 74: 00
2020-02-11_18_16_16_127: EngineState: 76: 00
2020-02-11_18_16_16_128: EngineState: 78: 00
2020-02-11_18_16_16_132: EngineState: 7A: 00
2020-02-11_18_16_16_133: EngineState: 7C: 00
2020-02-11_18_16_16_133: EngineState: 7E: 00
2020-02-11_18_16_16_134: EngineState: 00: FF
2020-02-11_18_16_16_134: EngineState: 02: 3F
2020-02-11_18_16_16_135: EngineState: 04: 3F
2020-02-11_18_16_16_135: EngineState: 06: 30
2020-02-11_18_16_16_135: EngineState: 08: 3F
2020-02-11_18_16_16_136: EngineState: 0A: 3F
2020-02-11_18_16_16_191: EngineState: 0C: 00
2020-02-11_18_16_16_192: EngineState: 0E: 00
2020-02-11_18_16_16_193: EngineState: 10: 00
2020-02-11_18_16_16_193: EngineState: 12: 00
2020-02-11_18_16_16_196: EngineState: 14: 00
2020-02-11_18_16_16_196: EngineState: 16: 00
2020-02-11_18_16_16_197: EngineState: 18: 00
2020-02-11_18_16_16_197: EngineState: 1A: 00
2020-02-11_18_16_16_198: EngineState: 1C: 09
2020-02-11_18_16_16_198: EngineState: 1E: A4
2020-02-11_18_16_16_201: EngineState: 20: 0D
2020-02-11_18_16_16_201: EngineState: 22: 00
2020-02-11_18_16_16_202: EngineState: 24: 00
2020-02-11_18_16_16_202: EngineState: 26: 00
2020-02-11_18_16_16_203: EngineState: 28: 01
2020-02-11_18_16_16_203: EngineState: 2A: 10
2020-02-11_18_16_16_203: EngineState: 2C: 0A
2020-02-11_18_16_16_204: EngineState: 2E: 3F
2020-02-11_18_16_16_204: EngineState: 30: 40
2020-02-11_18_16_16_222: EngineState: 32: 10
2020-02-11_18_16_16_222: EngineState: 34: 0F
2020-02-11_18_16_16_223: EngineState: 36: FF
2020-02-11_18_16_16_223: EngineState: 38: FF
2020-02-11_18_16_16_224: EngineState: 3A: FF
2020-02-11_18_16_16_224: EngineState: 3C: 0F
2020-02-11_18_16_16_250: EngineState: 3E: FF
2020-02-11_18_16_16_252: EngineState: 40: 77
2020-02-11_18_16_16_268: EngineState: 42: F7
2020-02-11_18_16_16_268: EngineState: 44: 77
2020-02-11_18_16_16_315: EngineState: 46: 47
2020-02-11_18_16_16_315: EngineState: 48: 1B
2020-02-11_18_16_16_316: EngineState: 4A: 00
2020-02-11_18_16_16_316: EngineState: 4C: 00
2020-02-11_18_16_16_316: EngineState: 4E: 00
2020-02-11_18_16_16_317: EngineState: 50: 00
2020-02-11_18_16_16_317: EngineState: 52: 00
2020-02-11_18_16_16_318: EngineState: 54: 00
2020-02-11_18_16_16_318: EngineState: 56: 00
2020-02-11_18_16_16_318: EngineState: 58: 00
2020-02-11_18_16_16_319: EngineState: 5A: 00
2020-02-11_18_16_16_319: EngineState: 5C: 00
2020-02-11_18_16_16_320: EngineState: 5E: 00
2020-02-11_18_16_16_320: EngineState: 60: 00
2020-02-11_18_16_16_320: EngineState: 62: 00
2020-02-11_18_16_16_321: EngineState: 64: 00
2020-02-11_18_16_16_321: EngineState: 66: 00
2020-02-11_18_16_16_322: EngineState: 68: 00
2020-02-11_18_16_16_322: EngineState: 6A: 00
2020-02-11_18_16_16_322: EngineState: 6C: 00
2020-02-11_18_16_16_323: EngineState: 6E: 00
2020-02-11_18_16_16_323: EngineState: 70: 00
2020-02-11_18_16_16_323: EngineState: 72: 00
2020-02-11_18_16_16_324: EngineState: 74: 00
2020-02-11_18_16_16_324: EngineState: 76: 00
2020-02-11_18_16_16_325: EngineState: 6A: 60
2020-02-11_18_16_16_325: EngineState: 7A: 00

dron0gus, 2020-02-12:
Due to bug address (first column) should be divided by 2. Also MSB is lost, so second half of table +0x40.
Address is taken from TLE8888 reply.
Differences are:
1. WD - this is expected.
2. OEConfig* registers are zeroed in first case. While these registers were set to 0xff on init.

dron0gus, 2020-02-12:
"Additional a dedicated output enable bit for each output is defined (see register OEConfig0 to OEConfig3) to avoid uncontrolled repetitive switching in failure case. These enable bits are reset by the protection function of each channel and block switch on of the channels. The bits could not be set if a protection function is active.
With setting the central enable bit to “1” all dedicated output enable bits are set to “1” (if no protection
function is active) and all channels are enabled and can be controlled according their configuration."

"Procedure to switch on after failure condition occurred:
• Read out of diagnosis bits
• Second read out to verify that the failure conditions are not remaining
• Set of the dedicated output enable bit of the affected channel if the diagnosis bit is not active anymore
• Switch on of the channel"

rusefi, 2020-02-12:
New data output, dump of registers at the end of tle8888SpiStartupExchange:

2020-02-12_08_33_13_548: EngineState: register: data
2020-02-12_08_33_13_549: EngineState: 00: 00
2020-02-12_08_33_13_549: EngineState: 00: 00
2020-02-12_08_33_13_550: EngineState: 01: 00
2020-02-12_08_33_13_550: EngineState: 02: 00
2020-02-12_08_33_13_550: EngineState: 03: 00
2020-02-12_08_33_13_551: EngineState: 04: 00
2020-02-12_08_33_13_551: EngineState: 05: 00
2020-02-12_08_33_13_561: EngineState: 06: 00
2020-02-12_08_33_13_562: EngineState: 07: 00
2020-02-12_08_33_13_562: EngineState: 08: 00
2020-02-12_08_33_13_562: EngineState: 09: 00
2020-02-12_08_33_13_563: EngineState: 0A: 00
2020-02-12_08_33_13_572: EngineState: 0B: 00
2020-02-12_08_33_13_573: EngineState: 0C: 00
2020-02-12_08_33_13_574: EngineState: 0D: 00
2020-02-12_08_33_13_574: EngineState: 0E: 00
2020-02-12_08_33_13_575: EngineState: 0F: 00
2020-02-12_08_33_13_577: EngineState: 10: 00
2020-02-12_08_33_13_578: EngineState: 11: 00
2020-02-12_08_33_13_578: EngineState: 12: 00
2020-02-12_08_33_13_580: EngineState: 13: 00
2020-02-12_08_33_13_581: EngineState: 14: 00
2020-02-12_08_33_13_582: EngineState: 15: 00
2020-02-12_08_33_13_583: EngineState: 16: 00
2020-02-12_08_33_13_584: EngineState: 17: 00
2020-02-12_08_33_13_585: EngineState: 18: 00
2020-02-12_08_33_13_585: EngineState: 19: 00
2020-02-12_08_33_13_588: EngineState: 1A: 00
2020-02-12_08_33_13_588: EngineState: 1B: 00
2020-02-12_08_33_13_589: EngineState: 1C: 00
2020-02-12_08_33_13_590: EngineState: 1D: 00
2020-02-12_08_33_13_592: EngineState: 1E: 00
2020-02-12_08_33_13_592: EngineState: 1F: 00
2020-02-12_08_33_13_593: EngineState: 20: 02
2020-02-12_08_33_13_596: EngineState: 21: 00
2020-02-12_08_33_13_596: EngineState: 22: 00
2020-02-12_08_33_13_598: EngineState: 23: 00
2020-02-12_08_33_13_598: EngineState: 24: 00
2020-02-12_08_33_13_600: EngineState: 25: 00
2020-02-12_08_33_13_601: EngineState: 26: AA
2020-02-12_08_33_13_602: EngineState: 27: AA
2020-02-12_08_33_13_603: EngineState: 28: AA
2020-02-12_08_33_13_603: EngineState: 29: AA
2020-02-12_08_33_13_605: EngineState: 2A: 00
2020-02-12_08_33_13_605: EngineState: 2B: 82
2020-02-12_08_33_13_607: EngineState: 2C: 00
2020-02-12_08_33_13_608: EngineState: 2D: 00
2020-02-12_08_33_13_610: EngineState: 2E: 00
2020-02-12_08_33_13_610: EngineState: 2F: 00
2020-02-12_08_33_13_612: EngineState: 30: 00
2020-02-12_08_33_13_613: EngineState: 31: 9C
2020-02-12_08_33_13_613: EngineState: 32: C2
2020-02-12_08_33_13_615: EngineState: 33: 00
2020-02-12_08_33_13_616: EngineState: 34: 09
2020-02-12_08_33_13_618: EngineState: 35: 40
2020-02-12_08_33_13_618: EngineState: 36: 30
2020-02-12_08_33_13_620: EngineState: 37: 30
2020-02-12_08_33_13_620: EngineState: 38: 30
2020-02-12_08_33_13_622: EngineState: 39: 30
2020-02-12_08_33_13_623: EngineState: 3A: 08
2020-02-12_08_33_13_623: EngineState: 3B: 09
2020-02-12_08_33_13_625: EngineState: 3C: 00
2020-02-12_08_33_13_626: EngineState: 3D: 00
2020-02-12_08_33_13_627: EngineState: 3E: 00
2020-02-12_08_33_13_628: EngineState: 3F: 00
2020-02-12_08_33_13_629: EngineState: 40: FF
2020-02-12_08_33_13_630: EngineState: 41: 3F
2020-02-12_08_33_13_630: EngineState: 42: 3F
2020-02-12_08_33_13_630: EngineState: 43: 30
2020-02-12_08_33_13_631: EngineState: 44: 3F
2020-02-12_08_33_13_631: EngineState: 45: 3F
2020-02-12_08_33_13_632: EngineState: 46: 00
2020-02-12_08_33_13_632: EngineState: 47: 00
2020-02-12_08_33_13_632: EngineState: 48: 00
2020-02-12_08_33_13_633: EngineState: 49: 00
2020-02-12_08_33_13_633: EngineState: 4A: 00
2020-02-12_08_33_13_633: EngineState: 4B: 00
2020-02-12_08_33_13_634: EngineState: 4C: 00
2020-02-12_08_33_13_634: EngineState: 4D: 00
2020-02-12_08_33_13_635: EngineState: 4E: 09
2020-02-12_08_33_13_635: EngineState: 4F: A4
2020-02-12_08_33_13_635: EngineState: 50: 0D
2020-02-12_08_33_13_636: EngineState: 51: 00
2020-02-12_08_33_13_636: EngineState: 52: 00
2020-02-12_08_33_13_637: EngineState: 53: 00
2020-02-12_08_33_13_637: EngineState: 54: 01
2020-02-12_08_33_13_637: EngineState: 55: 10
2020-02-12_08_33_13_638: EngineState: 56: 0A
2020-02-12_08_33_13_638: EngineState: 57: 3F
2020-02-12_08_33_13_638: EngineState: 58: 40
2020-02-12_08_33_13_639: EngineState: 59: 10
2020-02-12_08_33_13_639: EngineState: 5A: 0F
2020-02-12_08_33_13_639: EngineState: 5B: 00
2020-02-12_08_33_13_640: EngineState: 5C: 00
2020-02-12_08_33_13_640: EngineState: 5D: 00
2020-02-12_08_33_13_641: EngineState: 5E: 00
2020-02-12_08_33_13_641: EngineState: 5F: FF
2020-02-12_08_33_13_641: EngineState: 60: 77
2020-02-12_08_33_13_642: EngineState: 61: F7
2020-02-12_08_33_13_642: EngineState: 62: 77
2020-02-12_08_33_13_642: EngineState: 63: 47
2020-02-12_08_33_13_643: EngineState: 64: 1B
2020-02-12_08_33_13_643: EngineState: 65: 00
2020-02-12_08_33_13_644: EngineState: 66: 00
2020-02-12_08_33_13_644: EngineState: 67: 00
2020-02-12_08_33_13_644: EngineState: 68: 00
2020-02-12_08_33_13_645: EngineState: 69: 00
2020-02-12_08_33_13_645: EngineState: 6A: 00
2020-02-12_08_33_13_646: EngineState: 6B: 00
2020-02-12_08_33_13_646: EngineState: 6C: 00
2020-02-12_08_33_13_646: EngineState: 6D: 00
2020-02-12_08_33_13_647: EngineState: 6E: 00
2020-02-12_08_33_13_647: EngineState: 6F: 00
2020-02-12_08_33_13_648: EngineState: 70: 00
2020-02-12_08_33_13_648: EngineState: 71: 00
2020-02-12_08_33_13_649: EngineState: 72: 00
2020-02-12_08_33_13_649: EngineState: 73: 00
2020-02-12_08_33_13_650: EngineState: 74: 00
2020-02-12_08_33_13_650: EngineState: 75: 00
2020-02-12_08_33_13_650: EngineState: 76: 00
2020-02-12_08_33_13_651: EngineState: 77: 00
2020-02-12_08_33_13_651: EngineState: 78: 00
2020-02-12_08_33_13_652: EngineState: 79: 00
2020-02-12_08_33_13_652: EngineState: 7A: 00
2020-02-12_08_33_13_652: EngineState: 7B: 00
2020-02-12_08_33_13_653: EngineState: 7C: 00
2020-02-12_08_33_13_653: EngineState: 7D: 00

rusefi, 2020-02-12:
Same dump a minute later

2020-02-12_08_35_28_233: EngineState: confirmation_tle8888:7
2020-02-12_08_35_28_234: EngineState: register: data
2020-02-12_08_35_28_234: EngineState: 00: 00
2020-02-12_08_35_28_234: EngineState: 00: 00
2020-02-12_08_35_28_234: EngineState: 01: 00
2020-02-12_08_35_28_235: EngineState: 02: 00
2020-02-12_08_35_28_235: EngineState: 03: 00
2020-02-12_08_35_28_235: EngineState: 04: 00
2020-02-12_08_35_28_235: EngineState: 05: 00
2020-02-12_08_35_28_245: EngineState: 06: 00
2020-02-12_08_35_28_248: EngineState: 07: 00
2020-02-12_08_35_28_248: EngineState: 08: 00
2020-02-12_08_35_28_248: EngineState: 09: 00
2020-02-12_08_35_28_249: EngineState: 0A: 00
2020-02-12_08_35_28_264: EngineState: 0B: 00
2020-02-12_08_35_28_265: EngineState: 0C: 00
2020-02-12_08_35_28_266: EngineState: 0D: 00
2020-02-12_08_35_28_266: EngineState: 0E: 00
2020-02-12_08_35_28_268: EngineState: 0F: 00
2020-02-12_08_35_28_268: EngineState: 10: 00
2020-02-12_08_35_28_269: EngineState: 11: 00
2020-02-12_08_35_28_272: EngineState: 12: 00
2020-02-12_08_35_28_272: EngineState: 13: 00
2020-02-12_08_35_28_273: EngineState: 14: 00
2020-02-12_08_35_28_274: EngineState: 15: 00
2020-02-12_08_35_28_274: EngineState: 16: 00
2020-02-12_08_35_28_276: EngineState: 17: 00
2020-02-12_08_35_28_276: EngineState: 18: 00
2020-02-12_08_35_28_277: EngineState: 19: 00
2020-02-12_08_35_28_278: EngineState: 1A: 00
2020-02-12_08_35_28_278: EngineState: 1B: 00
2020-02-12_08_35_28_279: EngineState: 1C: 00
2020-02-12_08_35_28_281: EngineState: 1D: 00
2020-02-12_08_35_28_281: EngineState: 1E: 00
2020-02-12_08_35_28_288: EngineState: 1F: 00
2020-02-12_08_35_28_288: EngineState: 20: 02
2020-02-12_08_35_28_288: EngineState: 21: 00
2020-02-12_08_35_28_289: EngineState: 22: 00
2020-02-12_08_35_28_290: EngineState: 23: 00
2020-02-12_08_35_28_290: EngineState: 24: 00
2020-02-12_08_35_28_291: EngineState: 25: 00
2020-02-12_08_35_28_291: EngineState: 26: AA
2020-02-12_08_35_28_292: EngineState: 27: AA
2020-02-12_08_35_28_292: EngineState: 28: AA
2020-02-12_08_35_28_293: EngineState: 29: AA
2020-02-12_08_35_28_294: EngineState: 2A: 00
2020-02-12_08_35_28_320: EngineState: 35: 40
2020-02-12_08_35_28_321: EngineState: 35: 40
2020-02-12_08_35_28_321: EngineState: 2D: 00
2020-02-12_08_35_28_322: EngineState: 2E: 00
2020-02-12_08_35_28_322: EngineState: 2F: 00
2020-02-12_08_35_28_322: EngineState: 30: 00
2020-02-12_08_35_28_323: EngineState: 31: 98
2020-02-12_08_35_28_323: EngineState: 32: 02
2020-02-12_08_35_28_323: EngineState: 33: 01
2020-02-12_08_35_28_324: EngineState: 34: 19
2020-02-12_08_35_28_324: EngineState: 35: 40
2020-02-12_08_35_28_324: EngineState: 36: 00
2020-02-12_08_35_28_354: EngineState: 37: 00
2020-02-12_08_35_28_354: EngineState: 38: 2C
2020-02-12_08_35_28_354: EngineState: 39: 00
2020-02-12_08_35_28_355: EngineState: 3A: 80
2020-02-12_08_35_28_355: EngineState: 3B: 00
2020-02-12_08_35_28_355: EngineState: 3C: 00
2020-02-12_08_35_28_356: EngineState: 3D: 00
2020-02-12_08_35_28_356: EngineState: 3E: 00
2020-02-12_08_35_28_357: EngineState: 3F: 00
2020-02-12_08_35_28_357: EngineState: 40: FF
2020-02-12_08_35_28_357: EngineState: 41: 3F
2020-02-12_08_35_28_358: EngineState: 42: 3F
2020-02-12_08_35_28_358: EngineState: 43: 30
2020-02-12_08_35_28_359: EngineState: 44: 3F
2020-02-12_08_35_28_359: EngineState: 45: 3F
2020-02-12_08_35_28_359: EngineState: 46: 00
2020-02-12_08_35_28_360: EngineState: 47: 00
2020-02-12_08_35_28_360: EngineState: 48: 00
2020-02-12_08_35_28_360: EngineState: 49: 00
2020-02-12_08_35_28_361: EngineState: 4A: 00
2020-02-12_08_35_28_361: EngineState: 4B: 00
2020-02-12_08_35_28_361: EngineState: 4C: 00
2020-02-12_08_35_28_362: EngineState: 4D: 00
2020-02-12_08_35_28_362: EngineState: 4E: 09
2020-02-12_08_35_28_362: EngineState: 4F: A4
2020-02-12_08_35_28_363: EngineState: 50: 0D
2020-02-12_08_35_28_363: EngineState: 51: 00
2020-02-12_08_35_28_363: EngineState: 52: 00
2020-02-12_08_35_28_364: EngineState: 53: 00
2020-02-12_08_35_28_364: EngineState: 54: 01
2020-02-12_08_35_28_364: EngineState: 55: 10
2020-02-12_08_35_28_365: EngineState: 56: 0A
2020-02-12_08_35_28_365: EngineState: 57: 3F
2020-02-12_08_35_28_365: EngineState: 58: 40
2020-02-12_08_35_28_366: EngineState: 59: 10
2020-02-12_08_35_28_366: EngineState: 5A: 0F
2020-02-12_08_35_28_367: EngineState: 5B: 00
2020-02-12_08_35_28_367: EngineState: 5C: 00
2020-02-12_08_35_28_367: EngineState: 5D: 00
2020-02-12_08_35_28_368: EngineState: 5E: 00
2020-02-12_08_35_28_368: EngineState: 5F: FF
2020-02-12_08_35_28_368: EngineState: 60: 77
2020-02-12_08_35_28_368: EngineState: 61: F7
2020-02-12_08_35_28_369: EngineState: 62: 77
2020-02-12_08_35_28_369: EngineState: 63: 47
2020-02-12_08_35_28_369: EngineState: 64: 1B
2020-02-12_08_35_28_370: EngineState: 65: 00
2020-02-12_08_35_28_370: EngineState: 66: 00
2020-02-12_08_35_28_370: EngineState: 67: 00
2020-02-12_08_35_28_371: EngineState: 68: 00
2020-02-12_08_35_28_371: EngineState: 69: 00
2020-02-12_08_35_28_372: EngineState: 6A: 00
2020-02-12_08_35_28_372: EngineState: 6B: 00
2020-02-12_08_35_28_373: EngineState: 6C: 00
2020-02-12_08_35_28_373: EngineState: 6D: 00
2020-02-12_08_35_28_373: EngineState: 35: 40
2020-02-12_08_35_28_374: EngineState: 35: 40
2020-02-12_08_35_28_374: EngineState: 70: 00
2020-02-12_08_35_28_374: EngineState: 71: 00
2020-02-12_08_35_28_375: EngineState: 72: 00
2020-02-12_08_35_28_375: EngineState: 73: 00
2020-02-12_08_35_28_376: EngineState: 74: 00
2020-02-12_08_35_28_376: EngineState: 75: 00
2020-02-12_08_35_28_376: EngineState: 76: 00
2020-02-12_08_35_28_377: EngineState: 77: 00
2020-02-12_08_35_28_377: EngineState: 78: 00
2020-02-12_08_35_28_377: EngineState: 79: 00
2020-02-12_08_35_28_378: EngineState: 7A: 00
2020-02-12_08_35_28_378: EngineState: 7B: 00
2020-02-12_08_35_28_378: EngineState: 7C: 00
2020-02-12_08_35_28_379: EngineState: 7D: 20

rusefi, 2020-02-12:
New data output, -2QK 0.4.5 dump of registers at the end of tle8888SpiStartupExchange:
2020-02-12_08_53_11_433: EngineState: register: data
2020-02-12_08_53_11_454: EngineState: 00: 00
2020-02-12_08_53_11_455: EngineState: 00: 00
2020-02-12_08_53_11_455: EngineState: 01: 00
2020-02-12_08_53_11_455: EngineState: 02: 00
2020-02-12_08_53_11_456: EngineState: 03: 00
2020-02-12_08_53_11_456: EngineState: 04: 00
2020-02-12_08_53_11_457: EngineState: 05: 00
2020-02-12_08_53_11_457: EngineState: 06: 00
2020-02-12_08_53_11_457: EngineState: 07: 00
2020-02-12_08_53_11_458: EngineState: 08: 00
2020-02-12_08_53_11_458: EngineState: 09: 00
2020-02-12_08_53_11_458: EngineState: 0A: 00
2020-02-12_08_53_11_459: EngineState: 0B: 00
2020-02-12_08_53_11_459: EngineState: 0C: 00
2020-02-12_08_53_11_460: EngineState: 0D: 00
2020-02-12_08_53_11_460: EngineState: 0E: 00
2020-02-12_08_53_11_460: EngineState: 0F: 00
2020-02-12_08_53_11_461: EngineState: 10: 00
2020-02-12_08_53_11_461: EngineState: 11: 00
2020-02-12_08_53_11_461: EngineState: 12: 00
2020-02-12_08_53_11_462: EngineState: 13: 00
2020-02-12_08_53_11_462: EngineState: 14: 00
2020-02-12_08_53_11_462: EngineState: 15: 00
2020-02-12_08_53_11_463: EngineState: 16: 00
2020-02-12_08_53_11_463: EngineState: 17: 00
2020-02-12_08_53_11_464: EngineState: 18: 00
2020-02-12_08_53_11_464: EngineState: 19: 00
2020-02-12_08_53_11_464: EngineState: 1A: 00
2020-02-12_08_53_11_465: EngineState: 1B: 00
2020-02-12_08_53_11_465: EngineState: 1C: 00
2020-02-12_08_53_11_465: EngineState: 1D: 00
2020-02-12_08_53_11_466: EngineState: 1E: 00
2020-02-12_08_53_11_466: EngineState: 1F: 00
2020-02-12_08_53_11_466: EngineState: 20: 02
2020-02-12_08_53_11_467: EngineState: 21: 00
2020-02-12_08_53_11_467: EngineState: 22: 00
2020-02-12_08_53_11_467: EngineState: 23: 00
2020-02-12_08_53_11_468: EngineState: 24: 00
2020-02-12_08_53_11_468: EngineState: 25: 00
2020-02-12_08_53_11_469: EngineState: 26: AA
2020-02-12_08_53_11_469: EngineState: 27: AA
2020-02-12_08_53_11_469: EngineState: 28: AA
2020-02-12_08_53_11_469: EngineState: 29: AA
2020-02-12_08_53_11_470: EngineState: 2A: 00
2020-02-12_08_53_11_470: EngineState: 2B: 82
2020-02-12_08_53_11_471: EngineState: 2C: 00
2020-02-12_08_53_11_471: EngineState: 2D: 00
2020-02-12_08_53_11_472: EngineState: 2E: 00
2020-02-12_08_53_11_472: EngineState: 2F: 00
2020-02-12_08_53_11_472: EngineState: 30: 00
2020-02-12_08_53_11_473: EngineState: 31: D7
2020-02-12_08_53_11_473: EngineState: 32: 45
2020-02-12_08_53_11_473: EngineState: 33: 00
2020-02-12_08_53_11_474: EngineState: 34: 19
2020-02-12_08_53_11_474: EngineState: 35: 40
2020-02-12_08_53_11_474: EngineState: 36: 30
2020-02-12_08_53_11_475: EngineState: 37: 30
2020-02-12_08_53_11_475: EngineState: 38: 30
2020-02-12_08_53_11_475: EngineState: 39: 30
2020-02-12_08_53_11_476: EngineState: 3A: 00
2020-02-12_08_53_11_476: EngineState: 3B: 00
2020-02-12_08_53_11_476: EngineState: 3C: 00
2020-02-12_08_53_11_477: EngineState: 3D: 00
2020-02-12_08_53_11_477: EngineState: 3E: 00
2020-02-12_08_53_11_477: EngineState: 3F: 00
2020-02-12_08_53_11_481: EngineState: 40: FF
2020-02-12_08_53_11_512: EngineState: 41: 3F
2020-02-12_08_53_11_512: EngineState: 42: 3F
2020-02-12_08_53_11_513: EngineState: 43: 30
2020-02-12_08_53_11_513: EngineState: 44: 3F
2020-02-12_08_53_11_513: EngineState: 45: 3F
2020-02-12_08_53_11_514: EngineState: 46: 00
2020-02-12_08_53_11_514: EngineState: 47: 00
2020-02-12_08_53_11_514: EngineState: 48: 00
2020-02-12_08_53_11_514: EngineState: 49: 00
2020-02-12_08_53_11_515: EngineState: 4A: 00
2020-02-12_08_53_11_515: EngineState: 4B: 00
2020-02-12_08_53_11_515: EngineState: 4C: 00
2020-02-12_08_53_11_516: EngineState: 4D: 00
2020-02-12_08_53_11_516: EngineState: 4E: 09
2020-02-12_08_53_11_517: EngineState: 4F: A4
2020-02-12_08_53_11_517: EngineState: 50: 0D
2020-02-12_08_53_11_518: EngineState: 51: 00
2020-02-12_08_53_11_519: EngineState: 52: 00
2020-02-12_08_53_11_519: EngineState: 53: 00
2020-02-12_08_53_11_520: EngineState: 54: 01
2020-02-12_08_53_11_520: EngineState: 55: 10
2020-02-12_08_53_11_520: EngineState: 56: 0A
2020-02-12_08_53_11_521: EngineState: 57: 3F
2020-02-12_08_53_11_521: EngineState: 58: 40
2020-02-12_08_53_11_521: EngineState: 59: 10
2020-02-12_08_53_11_522: EngineState: 5A: 0F
2020-02-12_08_53_11_522: EngineState: 5B: FF
2020-02-12_08_53_11_522: EngineState: 5C: FF
2020-02-12_08_53_11_523: EngineState: 5D: FF
2020-02-12_08_53_11_523: EngineState: 5E: 0F
2020-02-12_08_53_11_523: EngineState: 5F: FF
2020-02-12_08_53_11_524: EngineState: 60: 77
2020-02-12_08_53_11_524: EngineState: 61: F7
2020-02-12_08_53_11_524: EngineState: 62: 77
2020-02-12_08_53_11_525: EngineState: 63: 47
2020-02-12_08_53_11_525: EngineState: 64: 1B
2020-02-12_08_53_11_525: EngineState: 65: 00
2020-02-12_08_53_11_526: EngineState: 66: 00
2020-02-12_08_53_11_526: EngineState: 67: 00
2020-02-12_08_53_11_526: EngineState: 68: 00
2020-02-12_08_53_11_527: EngineState: 69: 00
2020-02-12_08_53_11_527: EngineState: 6A: 00
2020-02-12_08_53_11_527: EngineState: 6B: 00
2020-02-12_08_53_11_528: EngineState: 6C: 00
2020-02-12_08_53_11_528: EngineState: 6D: 00
2020-02-12_08_53_11_528: EngineState: 6E: 00
2020-02-12_08_53_11_529: EngineState: 6F: 00
2020-02-12_08_53_11_529: EngineState: 70: 00
2020-02-12_08_53_11_529: EngineState: 71: 00
2020-02-12_08_53_11_529: EngineState: 72: 00
2020-02-12_08_53_11_530: EngineState: 73: 00
2020-02-12_08_53_11_530: EngineState: 74: 00
2020-02-12_08_53_11_530: EngineState: 75: 00
2020-02-12_08_53_11_531: EngineState: 76: 00
2020-02-12_08_53_11_531: EngineState: 77: 00
2020-02-12_08_53_11_531: EngineState: 78: 00
2020-02-12_08_53_11_531: EngineState: 79: 00

rusefi, 2020-02-12:
Again -QK with disable diag patch

2020-02-12_08_58_54_111: EngineState: register: data
2020-02-12_08_58_54_112: EngineState: 00: 00
2020-02-12_08_58_54_113: EngineState: 00: 00
2020-02-12_08_58_54_113: EngineState: 01: 00
2020-02-12_08_58_54_113: EngineState: 02: 00
2020-02-12_08_58_54_113: EngineState: 03: 00
2020-02-12_08_58_54_114: EngineState: 04: 00
2020-02-12_08_58_54_115: EngineState: 05: 00
2020-02-12_08_58_54_115: EngineState: 06: 00
2020-02-12_08_58_54_116: EngineState: 07: 00
2020-02-12_08_58_54_123: EngineState: 08: 00
2020-02-12_08_58_54_124: EngineState: 09: 00
2020-02-12_08_58_54_124: EngineState: 0A: 00
2020-02-12_08_58_54_124: EngineState: 0B: 00
2020-02-12_08_58_54_125: EngineState: 0C: 00
2020-02-12_08_58_54_125: EngineState: 0D: 00
2020-02-12_08_58_54_125: EngineState: 0E: 00
2020-02-12_08_58_54_126: EngineState: 0F: 00
2020-02-12_08_58_54_126: EngineState: 10: 00
2020-02-12_08_58_54_127: EngineState: 11: 00
2020-02-12_08_58_54_127: EngineState: 12: 00
2020-02-12_08_58_54_127: EngineState: 13: 00
2020-02-12_08_58_54_128: EngineState: 14: 00
2020-02-12_08_58_54_128: EngineState: 15: 00
2020-02-12_08_58_54_128: EngineState: 16: 00
2020-02-12_08_58_54_129: EngineState: 17: 00
2020-02-12_08_58_54_130: EngineState: 18: 00
2020-02-12_08_58_54_131: EngineState: 19: 00
2020-02-12_08_58_54_131: EngineState: 1A: 00
2020-02-12_08_58_54_134: EngineState: 1B: 00
2020-02-12_08_58_54_134: EngineState: 1C: 00
2020-02-12_08_58_54_135: EngineState: 1D: 00
2020-02-12_08_58_54_137: EngineState: 1E: 00
2020-02-12_08_58_54_138: EngineState: 1F: 00
2020-02-12_08_58_54_140: EngineState: 20: 02
2020-02-12_08_58_54_141: EngineState: 21: 00
2020-02-12_08_58_54_142: EngineState: 22: 00
2020-02-12_08_58_54_143: EngineState: 23: 00
2020-02-12_08_58_54_144: EngineState: 24: 00
2020-02-12_08_58_54_146: EngineState: 25: 00
2020-02-12_08_58_54_146: EngineState: 26: AA
2020-02-12_08_58_54_148: EngineState: 27: AA
2020-02-12_08_58_54_167: EngineState: 28: AA
2020-02-12_08_58_54_167: EngineState: 29: AA
2020-02-12_08_58_54_168: EngineState: 2A: 00
2020-02-12_08_58_54_169: EngineState: 2B: 82
2020-02-12_08_58_54_169: EngineState: 2C: 00
2020-02-12_08_58_54_169: EngineState: 2D: 00
2020-02-12_08_58_54_170: EngineState: 2E: 00
2020-02-12_08_58_54_170: EngineState: 2F: 00
2020-02-12_08_58_54_170: EngineState: 30: 00
2020-02-12_08_58_54_172: EngineState: 31: EF
2020-02-12_08_58_54_173: EngineState: 32: 14
2020-02-12_08_58_54_173: EngineState: 33: 00
2020-02-12_08_58_54_175: EngineState: 34: 09
2020-02-12_08_58_54_175: EngineState: 35: 40
2020-02-12_08_58_54_177: EngineState: 36: 30
2020-02-12_08_58_54_178: EngineState: 37: 30
2020-02-12_08_58_54_180: EngineState: 38: 30
2020-02-12_08_58_54_181: EngineState: 39: 30
2020-02-12_08_58_54_183: EngineState: 3A: 08
2020-02-12_08_58_54_183: EngineState: 3B: 09
2020-02-12_08_58_54_183: EngineState: 3C: 00
2020-02-12_08_58_54_185: EngineState: 3D: 00
2020-02-12_08_58_54_186: EngineState: 3E: 00
2020-02-12_08_58_54_186: EngineState: 3F: 00
2020-02-12_08_58_54_188: EngineState: 40: 00
2020-02-12_08_58_54_189: EngineState: 41: 00
2020-02-12_08_58_54_190: EngineState: 42: 00
2020-02-12_08_58_54_190: EngineState: 43: 00
2020-02-12_08_58_54_192: EngineState: 44: 00
2020-02-12_08_58_54_192: EngineState: 45: 00
2020-02-12_08_58_54_194: EngineState: 46: 00
2020-02-12_08_58_54_194: EngineState: 47: 00
2020-02-12_08_58_54_196: EngineState: 48: 00
2020-02-12_08_58_54_196: EngineState: 49: 00
2020-02-12_08_58_54_198: EngineState: 4A: 00
2020-02-12_08_58_54_198: EngineState: 4B: 00
2020-02-12_08_58_54_200: EngineState: 4C: 00
2020-02-12_08_58_54_200: EngineState: 4D: 00
2020-02-12_08_58_54_202: EngineState: 4E: 09
2020-02-12_08_58_54_202: EngineState: 4F: A4
2020-02-12_08_58_54_204: EngineState: 50: 0D
2020-02-12_08_58_54_204: EngineState: 51: 00
2020-02-12_08_58_54_206: EngineState: 52: 00
2020-02-12_08_58_54_208: EngineState: 53: 00
2020-02-12_08_58_54_209: EngineState: 54: 01
2020-02-12_08_58_54_210: EngineState: 55: 10
2020-02-12_08_58_54_211: EngineState: 56: 0A
2020-02-12_08_58_54_213: EngineState: 57: 3F
2020-02-12_08_58_54_213: EngineState: 58: 40
2020-02-12_08_58_54_215: EngineState: 59: 10
2020-02-12_08_58_54_216: EngineState: 5A: 0F
2020-02-12_08_58_54_218: EngineState: 5B: 00
2020-02-12_08_58_54_218: EngineState: 5C: 00
2020-02-12_08_58_54_220: EngineState: 5D: 00
2020-02-12_08_58_54_220: EngineState: 5E: 00
2020-02-12_08_58_54_223: EngineState: 5F: FF
2020-02-12_08_58_54_223: EngineState: 60: 77
2020-02-12_08_58_54_225: EngineState: 61: F7
2020-02-12_08_58_54_227: EngineState: 62: 77
2020-02-12_08_58_54_227: EngineState: 63: 47
2020-02-12_08_58_54_228: EngineState: 64: 1B
2020-02-12_08_58_54_230: EngineState: 65: 00
2020-02-12_08_58_54_230: EngineState: 66: 00
2020-02-12_08_58_54_232: EngineState: 67: 00
2020-02-12_08_58_54_233: EngineState: 68: 00
2020-02-12_08_58_54_234: EngineState: 69: 00
2020-02-12_08_58_54_234: EngineState: 6A: 00
2020-02-12_08_58_54_236: EngineState: 6B: 00
2020-02-12_08_58_54_236: EngineState: 6C: 00
2020-02-12_08_58_54_239: EngineState: 6D: 00
2020-02-12_08_58_54_248: EngineState: 6E: 00
2020-02-12_08_58_54_249: EngineState: 6F: 00
2020-02-12_08_58_54_249: EngineState: 70: 00
2020-02-12_08_58_54_251: EngineState: 71: 00
2020-02-12_08_58_54_251: EngineState: 72: 00
2020-02-12_08_58_54_253: EngineState: 73: 00
2020-02-12_08_58_54_254: EngineState: 74: 00
2020-02-12_08_58_54_254: EngineState: 75: 00
2020-02-12_08_58_54_256: EngineState: 76: 00
2020-02-12_08_58_54_257: EngineState: 77: 00
2020-02-12_08_58_54_257: EngineState: 78: 00
2020-02-12_08_58_54_258: EngineState: 79: 00
2020-02-12_08_58_54_258: EngineState: 7A: 00
2020-02-12_08_58_54_258: EngineState: 7B: 00
2020-02-12_08_58_54_259: EngineState: 7C: 00
2020-02-12_08_58_54_259: EngineState: 7D: 00

rusefi, 2020-02-12:
Status update:
35: 40

"Global Output Enable Status Bit" is set after initialization as expected in both cases, with and without WD

but individual channels are not being enabled

with WD
2020-02-12_08_35_28_367: EngineState: 5B: 00
2020-02-12_08_35_28_367: EngineState: 5C: 00
2020-02-12_08_35_28_367: EngineState: 5D: 00

without WD

2020-02-12_08_53_11_522: EngineState: 5B: FF IMPORTANT difference
2020-02-12_08_53_11_522: EngineState: 5C: FF IMPORTANT difference
2020-02-12_08_53_11_523: EngineState: 5D: FF IMPORTANT difference

rusefi, 2020-02-13:
and now it works :)

closed 2020-02-13 by rusefi