rusefi/rusefi #922: Delayed A/C compressor switch

rusefi, 2019-09-08:
At the moment NA6 turns on A/C compressor right the moment the button is toggled

to get happy idle we need to bump idle the moment A/C switch is toggled but only turn on the actual compressor X ms later, X=~100ms

rusefi, 2019-09-09:
new FSIO function: "time_since_ac_on_switch" returns number of seconds since most recent switch toggle

rusefi, 2019-09-12:
@stefanst:

IDLE settings dialog now has second offset and minValue fields
FSIO dialog now has new funny useFSIO12ForIdleOffset and useFSIO13ForIdleMinValue checkboxes

the formulas would be
```ac_on_switch cfg_idleRpmPid_offset cfg_idleRpmPid2_offset if
ac_on_switch cfg_idleRpmPid_minValue cfg_idleRpmPid2_minValue if```

rusefillc, 2020-10-04:
@stefanst @ZHoob2004 do we still need this ticket?

closed 2022-08-31 by mck1117