rusefi/rusefi #925: instant RPM for idle control

rusefi, 2019-09-09:
@stefanst is wondering if one of the ways to improve idle control would be to use instance RPM value instead of current per-cycle RPM value

There is a chance that per-cycle RPM takes too long to refresh and we are loosing valuable time

stefanst, 2019-09-09:
Correct. On a 1.6l Miata engine, coming off of 2000rpm, free-revving, we have 3 full crankshaft revolutions from rpm>1000 (which is the target) to a stalled engine. The "I" factor has only 2-3 true iterations to adjust.

rusefi, 2019-09-09:
Reminder that we have "INSTANT_RPM" debug mode which allows to monitor that instant RPM value vs "regular" RPM which is currently used more often

mck1117, 2021-05-27:
We use instant RPM already for closed loop idle timing correction, which is the part that really matters.

OrchardPerformance, 2021-10-03:
@mck1117 - Can this be closed? Do you feel we have resolved this one?

mck1117, 2022-06-22:
this is well past fixed, and instant rpm wasn't even the problem.

closed 2022-06-22 by mck1117