rusefi/rusefi #927: Funny things on NB2 ETB drive log

rusefi, 2019-09-10:
TBD @mck1117

mck1117, 2019-09-10:
[image]

Observe blips to 0/100 on the "Idle Air Valve" channel.  The RPM isn't getting updated very often, so when it does get updated, it's a big step.  The stair steps in the above log snip are 847, 795, 741, 728, 751, etc, each step around 0.17 second apart.

The yellow channel, calculated in the log viewer (not a real channel), shows that the RPM is instantaneously in a single sample dropping by ~50, causing a large spike in the idle position output as a result of its D term.

mck1117, 2019-09-10:
There are a few things that could be improved here:

1) Calculate RPM more often than every engine revolution or cycle.  0.17 second is a long time, and is likely causing other issues because of the phase lag it introduces to sensing RPM.  This will help prevent the stepwise RPM behavior, instead giving a smoother, more continuous curve.  (This change also has the potential to make the engine considerably more responsive off-idle).

2) The D term likely isn't needed for the idle controller.  Without a D term, there is no opportunity for high frequency noise in the RPM calculation to make it out to the idle control hardware/ETB.  Idle control is also slow enough (and should be able to be tuned lazily enough) that the D term doesn't add any practical benefit to idle stability.

mck1117, 2021-01-03:
this will be part way to fixed by #2159 - by computing dRpm/dT only when the RPM changes, spikes like this are avoided.

mck1117, 2021-07-04:
this is long since fixed

closed 2021-07-04 by mck1117