rusefi/rusefi #936: misfire detection

rusefi, 2019-09-15:
Can we please detect broken injector or broken coil programmatically?

chuckwagoncomputing, 2021-01-05:
#925 #343 #2180

rusefillc, 2021-07-19:
See https://rusefi.com/forum/viewtopic.php?f=5&t=2071

chuckwagoncomputing, 2021-07-19:
FWIW, Ford's IDS software has an awesome cylinder contribution graph that has also been copied by other scan tool software. It's great that the ECU can say "cylinder n is misfiring", but being able to visualize the data is _awesome_.

rusefillc, 2021-07-19:
@chuckwagoncomputing any screenshots? :)

chuckwagoncomputing, 2021-07-19:
Even better, a video:
https://www.youtube.com/watch?v=G6WYnNlYuiQ
The video is a diesel, but the same thing exists for gas vehicles.

stefanst, 2021-07-19:
Trying to start doing this with a low-resolution Miata crank trigger may be a tall initial hurdle to jump. Wouldn't this be easier using a 60-2 or 36-2 or similar trigger?

thirstyone, 2021-07-19:
even with this config you can clearly see the difference in timing (I just looked through the second file in saleae - without injector, I believe, without a spark it would be even more obvious)

chuckwagoncomputing, 2021-07-19:
Doesn't the trigger decoder already calculate where it "expects" the tooth to be, versus where it is? A visualization of this data would be better than nothing. Perhaps as an overlay of the trigger diagram in Console?

rusefillc, 2021-07-19:
@chuckwagoncomputing we have a _proposal_ #2222 but only a proposal.

Our reference diagram at https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#mazda-miata-nb

rusefillc, 2022-01-11:
We have data and proposal at https://rusefi.com/forum/viewtopic.php?p=42922#p42922