rusefi/rusefi #943: ETB is sticky around zero and does not want to close all the way

rusefi, 2019-09-21:
See attached log

start car
pedal = 0%
IDLE control OFF

TPS does not close to 0% and RPM stays relatively high

once in a while if one touches pedal throttle would shut closed as expected, but PID does not seem to be closing it quickly with my current settings at least

See how RPM went down after throttle blip

[image]

rusefi, 2019-09-21:
[rus0921_163445_sticky_throttle_at_zero.zip](https://github.com/rusefi/rusefi/files/3639158/rus0921_163445_sticky_throttle_at_zero.zip)
[TunerStudioDebugPackage.zip](https://github.com/rusefi/rusefi/files/3639159/TunerStudioDebugPackage.zip)

mck1117, 2019-09-22:
Summary of discussion on Slack: this is *probably* a hardware issue (mechanical, not rusEfi).  @rusefi's Miata has the wrong throttle body installed, which is causing a considerable vacuum leak.

OrchardPerformance, 2020-06-01:
@mck1117 - Do you see this issue on your new ETB install? 
In not do we close this as a hardware issue?

mck1117, 2020-06-01:
this was a hardware issue, yeah

closed 2020-06-01 by rusefi