rusefi/rusefi #948: HardFault related to auto r = convertFromInputValue(inputValue);

rusefi, 2019-09-23:
I have a discovery with old and probably invalid configuration

I've flashed it with current firmware

First I get a fatal error
[image]

But we kind of expect to handle all fatals so that at least console is up and running and we can read fatal error message and fix configuration.

But next I get:
[image]

rusefi, 2019-09-23:
@mck1117 please advise

rusefi, 2019-09-23:
@mck1117 

[image]

rusefi, 2019-09-23:
@mck1117 attached are my _current_ .map and .elf files. NOT matching  affected tag

https://github.com/rusefi/rusefi/commit/344919a60b4ea2f3745e5fdfa786701e943ff82e

I hope this version is still affected. Let me know if it's not and I will roll back to https://github.com/rusefi/rusefi/commit/344919a60b4ea2f3745e5fdfa786701e943ff82e and compile it

mck1117, 2019-09-23:
The bug is that `ltoa_internal` would always write the next 11 bytes, even if the buffer wasn't that long (which it didn't know).

I've rewritten `ltoa_internal` so that it doesn't write any more than it needs to in the output buffer, using a stack buffer instead of the destination for intermediate scratch work.

PR #949 (2019-09-23): fix ltoa_internal: fix #948
Fixes #948
commit b4e176a9 (2019-09-23): fix ltoa_internal: fix #948 (#949)

* fix ltoa_internal

* fix hex
closed 2019-09-23 by rusefi