rusefi/rusefi #951: how do plan to use

rusefi, 2019-09-24:
Do we have a plan for IS25LP256D on MRE?

See m25q.h in ChibiOS
See hal_jesd216_flash.h in ChibiOS
See jesd216

rusefi, 2019-09-28:
See http://www.chibios.com/forum/viewtopic.php?f=38&t=3320&start=10

rusefi, 2019-09-29:
[image]

ChibiOS seems to have a driver for N25Q128 

[image]

rusefi, 2019-10-03:
"Changed the m25q -> s25f for use with Cypress S25FL116/132/162 Flash Devices."

rusefi, 2020-04-20:
See also https://github.com/vooon/chibios-flash
See also https://github.com/nimaltd/w25qxx

mck1117, 2022-06-23:
we don't, because we've stopped putting it on boards in lieu of SD card

closed 2022-06-23 by mck1117