rusefi/rusefi #952: Allow boards/mcus to set -fcpu= option

mck1117, 2019-09-24:
GCC emits slightly different code for `-fcpu=cortex-m4` and `-fcpu=cortex-m7` - mostly around instruction ordering to prevent pipeline stalls.

Here's a small example: https://godbolt.org/z/T4upjz

The `vmov.f32` and `vcvt.f32.s32` instructions got swapped to better utilize the pipeline, likely saving 1 or 2 cycles just by changing instruction order.

Currently the makefile blindly sets the compiler to build for `cortex-m4` on line 325.  CM4 is a fine default, but we should be able to override elsewhere when building for stm32f7 or other cortex-m7 chips.

dron0gus, 2019-12-24:
https://github.com/dron0gus/rusefi/commit/40a399f5594493aecdde96348a7f64b37c3731a1

mck1117, 2021-03-18:
this is done

closed 2021-03-18 by mck1117