rusefi/rusefi #954: absolute fuel pressure flow calculation

rusefi, 2019-09-24:
Initially rusEfi only supported "map referenced" flow rate - i.e. fuel rail pressure follows manifold pressure (vacuum line into fuel regulator is part of mechanical setup)

Some vehicles like US Miata NB2 have absolute fuel pressure setup - i.e. no vacuum line to regulator and constant fuel pressure.

We shall add "absolute fuel pressure" mode with fuel system pressure configuration setting.

rusefi, 2019-09-24:
http://performancefuelsystems.com/InjectorFlowRatesatDifferentPressures-TechCorner.htm

rusefi, 2019-10-30:
Nice write-up at https://rusefi.com/forum/viewtopic.php?f=5&t=1636

mck1117, 2021-03-28:
This mode is now working and set to the default on the NB miata tune.

closed 2021-03-28 by mck1117