rusefi/rusefi #958: human readbale error codes in TS

rusefi, 2019-09-28:
TS supports human readable error codes

```[PcVariables]     troubleList  = bits,   U08,   [0:2], "Ok", "O2 Sensor", "CLT Sensor", "IAT Sensor"
 indicator = { malFunc }, "No Codes", {Error: bitStringValue(troubleList, malFunc)},   green, black, red, black```

we shall use this for fatal errors at least

probably yet another code generator. maybe different enum numbers for error codes?

rusefi, 2019-11-04:
```[PcVariables]
troubleList  = bits,   U08,   [0:7], "Error: Pin already used", "Error: Pin already used", "Invalid CAN Pin","MAP out of range", "IAT out of range","CLT Temp out of range", "Warning: TPS Too Low","Warning: TPS Too High", "Warning: Trigger Noise","No cam signal", "Voltage out of range","Prosessor Fault", "ETB Rest position","Invalid baro pressure", "Invalid Baro Correction", "Trigger Sync Error", "Trigger Sync Warning", "Warning: No Engine Load", "Wrong Algo for MAP based timing" ,"Warning: No Engine Load", "FSIO Empty stack", "FSIO Underdefined", "FSIO undefined action", "FSIO unexpected Stack size", "FSIO unexpected action","FSIO Warning Parsing Formula", "FSIO Invalid Expression", "Interpolate Error", "Interpolate Error", "Interpolate Error", "Interpolate Error", "map not initialized",	"MAF sensor needed for current fuel algorithm", "Unexpected engine load parameter", "Coil Pin Not Assigned","Injector Pin Not Assigned","Unexpected Injection Mode","angle constraint violation in findTriggerPosition", "Unknown Firing Order", "Wrong Firing Order", "Unsupported IgnitionMode", "Unexpected engine type", "Invalid TPS Setting", "No Injection Pulse", "Negative Injection Pulse", "Zero Dwell", "Dwell Too Long", "Skipping stroke due to rpm", "Trigger decoding issue", "Oil Pressure sensor out of range", "Flash CRC failed", "Zero Cranking Fuel"

indicator = { recentErrorCode0 }, "No Error Codes", {bitStringValue(troubleList, recentErrorCode0)},   green, black, red, black 
indicator = { recentErrorCode1 }, "No Error Codes", {bitStringValue(troubleList, recentErrorCode1)},   green, black, red, black ```

rusefi, 2019-11-06:
Related/duplicate of #937?

mck1117, 2020-04-18:
is this done? we have critical error msg in ts

closed 2020-04-18 by rusefi