rusefi/rusefi #959: Many tests depend on working fuel math

mck1117, 2019-09-28:
Why does `misc.testRpmCalculator` fail if fueling math doesn't return the correct value?

rusefi, 2019-09-28:
because unit tests are pretty much really not unit tests but integration tests

mck1117, 2019-09-28:
that's exactly why I filed the bug

we need unit tests

mck1117, 2019-09-28:
The problem is that when you change something - completely unrelated tests start failing.

Unit tests should rely on mocks to test only one part of the system.

Integration tests can't rely on mocks, as they need to test the system as a whole.

rusefi, 2019-09-28:
I agree completely but in my opinion this is totally unrealistic.

mck1117, 2019-09-28:
Why is that unrealistic?

mck1117, 2019-09-28:
Let me ask the question that prompted this:

Suppose I change how some system works (clt/iat sensors, for example).  Unrelated tests start failing.  What do?

rusefi, 2019-09-28:
I believe that the benefits do not justify the investments. It is my opinion that while so little firmware progress is done, it would make no sense to spend resources on improving unit tests from current acceptable state to a better state.

mck1117, 2019-09-29:
But the cost of that right now is that progress is extremely difficult to make. Trying to make a change results in spending hours chasing down why that changed a test behavior.

rusefi, 2019-09-29:
I am open to PRs

closed 2020-12-25 by mck1117