rusefi/rusefi #964: validate flash size on start

rusefi, 2019-10-03:
firmware should dynamically inquire and assert flash size

rusefi, 2019-10-03:
See https://stm32f4-discovery.net/2014/12/library-44-stm32f4-unique-id-and-flash-size/

mck1117, 2019-10-08:
So as to trip an error if we try to read/write config past the end of flash?

rusefi, 2019-10-08:
Not just that but potentially execute past end of flash

mck1117, 2019-10-08:
Sure - but it's difficult to guarantee that all of the code required to detect/report that error gets located in the first N bytes of flash.  So long as we keep the full image below the minimum flash size of any MCU we support (512k, I believe), then we're safe on execution out of bounds.

mck1117, 2019-11-12:
We can also have efifeatures set an expected signature for the ECU - that will prevent against both "wrong size flash" and "wrong mcu" execution.

mck1117, 2021-11-29:
we now dynamically handle flash size on F7

closed 2021-11-29 by mck1117