rusefi/rusefi #965: With ETB active and high RPM 36-1 trigger loss of communication

rusefi, 2019-10-03:
36/1 VR + 1 CAM trigger setup
https://rusefi.com/forum/viewtopic.php?f=5&t=592&start=240#p34619

stm32f407 default firmware

with 3800 RPM we get microUSB/native real USB pausing communication

communication comes back once RPM get below

rusefi, 2019-10-03:
Looks like performance issue?

1) try much lower ETB frequency just to validate the theory - not 300Hz but 30Hz
2) try "no_assert" version of the firmware just to see if issue would move away to higher RPM

Rene325, 2019-10-03:
I tried the rusefi_no_asserts firmware and that helped a lot.
Unstabillity in control and comms above 7500 RPM
I tried both 30 and 300Hz PWM Freq. that does not make any difference

rusefi, 2019-10-17:
Probably related to #973

Could be fixed by #973

rusefi, 2019-11-04:
Still an issue as reported at https://rusefi.com/forum/viewtopic.php?f=5&t=1629#p34788

rusefi, 2020-06-19:
I am sure it's not an issue any more with all the progress @mck1117 has made

closed 2020-06-19 by rusefi