rusefi/rusefi #967: cooling fan to turn off in cranking

rusefi, 2019-10-06:
cooling fan should be off during cranking even if engine is hot

mck1117, 2019-10-08:
Maybe the logic should be
`fan = (clt > thresh) && (uptime > 10) && !cranking`

960, 2019-10-20:
If you wire your fan relay correct, this advanced feature are controlled by ignition switch

mck1117, 2019-10-21:
I've never seen an OEM cooling fan wired in any way that the ECU doesn't have full control (no ignition switch involvement).  Removing loads from the ignition switch is generally a good idea, and can significantly simplify wiring the car.

960, 2019-10-21:
You are joking, right?

You know the logic of your ignition switch?
In S-position, you should have a output who are not active.

In newer cars, the ecu have full control of it, but you can still wire it different.

It's not a load on the ignition switch at all, just the 85 wire to relay

mck1117, 2019-11-12:
> In newer cars, the ecu have full control of it, but you can still wire it different.

Exactly - so we should support that case too.

mck1117, 2019-11-12:
In other words, we should do our best to support all cases of "I cut the connectors off and wired in a new ECU without otherwise rewiring the car".  The truck that my V8 is from had the fan wired like I'm describing (no involvement of the ignition switch, full ECU control), and it was wired like that as early as 1997.

960, 2019-11-12:
I have never seen a fan running while cranking myself, they have always been at ACC  output of ignition switch.
You may be right it exists, I just haven't seen it yet.
Isn't this what you have your FSIO for? 
Cant you just make a polish formula?

mck1117, 2019-11-12:
I think you're misunderstanding my point.  OE ECUs that have full control of the fan disable the fan while cranking.  Via software. Which we don't currently support.  For example, my Focus doesn't even have an ignition switch.  Pressing the start/stop button asks the ECU and BCM to "please turn the car on" which does different things whether you're actually starting the engine or not.  For example, it also kills the headlights while cranking.

960, 2019-11-12:
I am aware of that. Still I havent seen any ECU control it, only BCM.

It's just like ACC on a old car. Normally the start-button are not connected to the ECU at all, just the BCM.

It turns off all heavy loads while cranking, and talks to the ecu via can(rpm etc).

So then we are no longer talking about the fan at all, but all heavy/unneeded loads.

mck1117, 2021-11-29:
we have a setting for this now

closed 2021-11-29 by mck1117