rusefi/rusefi #972: Measurement of CPU load

mck1117, 2019-10-11:
How busy is the CPU?

Are interrupts regularly waiting on one another?

Is our generous use of global locking causing issues?

rusefi, 2019-10-11:
ChibiOS has some mechanisms and rusEfi already has some custom mechanisms. Looks like better documentation of those is needed :)

mck1117, 2019-10-12:
We could easily generate traces convertible to view with this:
https://github.com/catapult-project/catapult/tree/master/tracing

rusefi, 2019-10-13:
@mck1117 please PR and document your tracing setup. What is the workflow exactly?

```ScopePerf perf(PE::MainTriggerCallback);``` is great but please elaborate a bit.

rusefi, 2019-11-04:
@mck1117 I am still very interested in this :)

mck1117, 2021-03-18:
99% covered by the perf tracing implementation

closed 2021-03-18 by mck1117