rusefi/rusefi #973: Both HandleFuel and PeriodicFastCallback call GetBaseFuel

mck1117, 2019-10-13:
Why do both of these compute the same thing? Should one be storing it for the other to use?

[image]

Top row is the trigger interrupt.  Bottom row is the periodic timer running PeriodicFastCallback.

mck1117, 2019-10-15:
Some napkin math: the trigger interrupt invocation, on a 60-2 wheel, at 6000 RPM, gets called 5800 times per second, representing around 14% CPU usage.

rusefi, 2019-10-15:
[image]

I can only think of acceleration enrichment as a reason?

mck1117, 2019-10-15:
A reasonable compromise may be:
1) remove it from `handleFuel`
2) increase `PeriodicFastCallback` to 200hz

mck1117, 2019-10-15:
That will only use up around 2% CPU for the `PeriodicFastCallback`, and will run it at a lower priority than it currently runs, to prevent starvation of the trigger decoder interrupt.

rusefi, 2019-10-16:
Prior to merging existing PR I would like to add more unit tests to ensure that functionally we would not be affected

rusefi, 2019-10-17:
Fixed by #979

closed 2019-10-17 by rusefi