rusefi/rusefi #974: MapAveragingCallback schedules all cylinders at once

mck1117, 2019-10-13:
[image]

Scheduling (2 * N) callbacks at once seems like a not great idea.  `MapAveragingTriggerCallback` alone took 26us on a release build, but only one per revolution.  This has significant potential to cause jitter.

mck1117, 2019-10-13:
And a corollary: should there be an API for scheduling a batch of events, without calling `DoExecute` per item (only call it once at the end)?

[image]

(this is is a zoom-in of one of the 16 enqueue events above)

mck1117, 2019-10-13:
Some napkin math suggests that the current logic on an F4 can't run a v8 over 4000 RPM without missing interrupts.

rusefi, 2019-10-13:
Great find, thank you!

rusefi, 2020-04-23:
Nick reported hanging self-stimulation at 6K with 60/2 once he has enabled one of the two tachometer implementations or something like that.

Maybe it was about this issue really?

mck1117, 2020-04-23:
Another consideration: We can't necessarily amortize the cost across multiple teeth, since some trigger patterns may not have more teeth, forcing our hand.  However, those triggers are also at less risk of having interrupts stomp on one another.

rusefillc, 2020-10-03:
@mck1117 poke :)

rusefillc, 2025-07-19:
@FDSoftware this is done, right?

FDSoftware, 2025-07-19:
> [@FDSoftware](https://github.com/FDSoftware) this is done, right?

yup, was https://github.com/rusefi/rusefi/pull/8099 / https://github.com/rusefi/rusefi/pull/8034