rusefi/rusefi #976: ADC sample time & rate

mck1117, 2019-10-14:
The ADC sample times & rates should probably be revisited.  480 samples likely isn't adding value over a shorter sample time, and the slow ADC could also be sampled more often.

mck1117, 2019-10-14:
ADC clock is 21MHz on F4, and 27MHz on F7

DonaldBecker, 2019-10-14:
The basic calculation is to get within a count at 2^N accuracy.

R_ain + R_adc = (k - 0.5) / (f_adc * C_adc * ln(2^N))

The STM32 manual targets a 1/4 LSB despite various noise and error sources swamping that number, and so uses 2^14 for a 12 bit ADC.

The STM32F40x datasheet states the following values
R_adc  = 6 Kohm (max)  Internal sampling switch resistance
C_adc = 5 pF (nominal)   Internal sampling capacitor

Sampling periods available are 3,  15,  28, 56,  84, 112, 144, 288

Note that the internal R_adc means that decreasing the external input impedance has quickly diminishing returns below 6 Kohms, and that the 3 tick sample is primarily useful with much slower clock rates.

DonaldBecker, 2019-10-19:
Plugging in the numbers for the STM32F407

R_ain + 6 Kohm = (k - 0.5) / (21 MHz * 5pF * 9.7)

  = (k - 0.5) / (1.018 * 10^-3)  or roughly  (k - 0.5)*1000

So we have the following useful settings:

A 15 cycle sample time supports a total resistance of about 15 Kohm, allowing an external resistance of 9 Kohm.  This suits the existing analog inputs, even with the somewhat faster F7 clocks.

A 28 cycle sample time allows an external resistance of about 20 Kohm.
A 56 cycle sample time allows an external resistance of about 50 Kohm.  This might be useful with
other hardware that uses unbuffered voltage divider inputs.

Since 50 KOhm is the maximum recommended, there is no reason to use a longer sampling time than 56 cycles with the current clock.  Using the 3 cycle sampling time would only provide accuracy if the internal sampling switch resistance is at the minimum side of the datasheet range (1.5 KOhm), something that cannot be relied upon.

rusefi, 2019-11-12:
@mck1117 do we need more changes here? what is the plan if any?

OrchardPerformance, 2020-06-01:
@mck1117 - nudge?

closed 2020-06-01 by rusefi

mck1117, 2021-07-20:
info about H7: https://community.st.com/s/question/0D50X0000BD2Etj/radc-value-for-stm32h753