rusefi/rusefi #986: 60/2 VW trigger has running issue on 19/10/19 snapshot

OrchardPerformance, 2019-10-19:
*I will add some more information later on including logs*

Using the 19/10/19 rusEFI bundle with a known good tune results in an initial start and then stall unless the throttle is held. As soon at the throttle is released the engine dies. 
Logs seem to show an odd short dip of crank RPM shortly before the engine dies.

This behavior is not seen on the older (28/08/19) snapshot. 

A-B-A test done to confirm the fault was being produced by the current firmware. 
I have several logs of it both working and not working.

rusefi, 2019-10-20:
Most recent changes: ADC sampling duration went down under #976

More probably cause of issue: untested on real hardware change to IAC startup procedure under #975

does idle seem to be working?

mck1117, 2019-10-21:
Can you try to grab an engine sniffer trace of the behavior? That may contain clues about what's going on.

I can test on my Bosch-derived 60-2 setup tonight.  I'm running firmware from October 16.  I have no idle valve though, so if that's related, I may or may not see the issue.

closed 2019-10-31 by rusefi

rusefi, 2019-10-31:
idle hardware was simply not initialized on start up, nothing to do with trigger

work-around would be to change settings in runtime - setting change might have actually start the hardware

commit 4d095d48 (2019-10-31): 60/2 VW trigger has running issue on 19/10/19 snapshot fix #986