rusefi/rusefi #988: 60/2 vs 60/2 VW mess

rusefi, 2019-10-30:
We have two versions of 60/2 trigger - "60/2" and "60/2 VW"

looks like this is due to misunderstanding of how VR works, one is theoretical nice 60/2 Hall while the other one is real 60/2 VR signal

See also https://github.com/rusefi/rusefi/issues/707

rusefi, 2019-10-30:
See https://github.com/rusefi/rusefi_documentation/raw/master/triggers/hall_vs_vr.png at https://rusefi.com/forum/viewtopic.php?f=2&t=1519

rusefi, 2019-10-30:
Matthew Kennedy 9:39 PM
a hall sensor engine will work fine with the VR config (ie, only decode one edge)
event every 6 degrees vs. event every 3 degrees isn't a difference, other than sucking up cpu cycles
new messages
whatever edge and type of sensor you have, if you see gap ratios of 1-1-1-2-2-1-1-1 instead of 1-1-1-3-1-1-1 then the polarity is wrong
russian 9:42 PM
hold on
do you mean
1-1-1-2-2-1-1-1 means polarity is wrong
and in case of wrong polarity, you will see 1-1-1-2-2-1-1-1?
Matthew Kennedy 9:42 PM
Yes

rusefi, 2019-11-16:
[image]
[image]

this is from my M73 confirming above opinion

OrchardPerformance, 2020-05-24:
Should these be referred to as 60-2 High gap and 60-2 Low gap.
Then it identifies the two traces by the physical feature of the wheels. 
I think one signal comes from each kind of trigger wheel gap.

OrchardPerformance, 2020-05-24:
[image]
Low gap

OrchardPerformance, 2020-05-24:
[image]
high gap

OrchardPerformance, 2020-05-24:
[image]

rusefillc, 2021-04-11:
Status update: we are still a bit confused :)

But at least we now have nicer wheel shapes documentation at

https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#vw-602
and
https://github.com/rusefi/rusefi/wiki/All-Supported-Triggers#602

mck1117, 2021-04-12:
Neither of those patterns matches the Renault pattern.  The renault pattern missing tooth looks similar, but the first tooth after the gap is longer.

rusefillc, 2021-06-27:
@mck1117 FYI

mck1117, 2021-06-27:
Nik's trigger looks like this:
[image]

mck1117, 2021-06-27:
My missing tooth (missing gap) looks like this:
[image]

mck1117, 2021-06-27:
As far as I can tell, these are the same, but one is inverted from the other.

mi-hol, 2023-04-02:
My take from above comments is:
- there is no real technical issue but just unclear documentation

Is this correct understanding?

rusefillc, 2023-04-02:
Not sure. Documentation implies user would manually locate documentation and understand it, I would prefer that we do smarter things like auto-detect the issue etc.

mi-hol, 2023-04-02:
> I would prefer that we do smarter things like auto-detect the issue etc.

While I understand and support this view, realistically [auto-detect trigger pattern](https://github.com/rusefi/rusefi/issues/418) seems a too big task considering the currently available resources to fix this within a reasonable amount of time. Auto-detection had apparently no activities since 2017!
A viable short term fix to move forward with https://github.com/rusefi/rusefi_documentation/issues/459 seems the approach with documentation only.
I'd be happy to be proven wrong though

rusefillc, 2024-04-21:
M73 with VW trigger

[image]

https://github.com/rusefi/rusefi/wiki/Trigger-Configuration-Guide#determine-correct-polarity says it's flipped

rusefillc, 2024-04-21:
same M73 engines now flipped wires

[image]

rusefillc, 2024-04-21:
https://rusefi.com/forum/viewtopic.php?p=49489#p49489

closed 2024-04-21 by rusefillc