rusefi/rusefi #996: Strange INJ1 signal behavior

andreika-git, 2019-11-02:
Setup:
- Deucalion (Kinetis-only)
- 60-2 "arduino-ish" simulator (workbench setup)
- ~350 RPM (cranking), simult.inj. mode
- INJ_OUT1 set to TLE6240_12
- PB4 pin used as additional debug output pin:
[image]

The problem is shown here:
[image]
The first signal is PB4 output, the second signal is TLE6240_12 output (slightly delayed and with small jitter, but will do), the 3rd signal is the trigger 'crank' input (the 'cam' input was also used).

The log is ok, no warnings or trigger errors while 'cranking' etc.:
[image]
Also no 'special' messages in the console.

The ini/settings, log and logicdata are attached.
[inj1_rpm350.zip](https://github.com/rusefi/rusefi/files/3800391/inj1_rpm350.zip)

andreika-git, 2019-11-02:
UPDATE: The problem seems to exist only for cranking mode (?). If I increase RPM to 1000+ (i.e. switch to 'running') and decrease back to 200-300 (still 'running', not 'cranking'?), then the INJ signal is ok:
[image]
But if I interrupt the trigger signal for a couple of seconds ("the engine stopped"), and restore it, then we're back to 'cranking', and INJ1 signal is wrong again (or even disappear (!), if RPM is too low, like <200)...

rusefi, 2019-11-04:
I wonder if this resembles https://github.com/rusefi/rusefi/issues/796

mck1117, 2019-11-05:
I remember a bug a while ago where very short injection durations could cause the injector phase to flip, sticking the injector open. I observed this during decel fuel cut flooding the engine with stuck open injector(s).  The on/off events were scheduled at the same time, so it was a coin flip about which one actually ran first.

andreika-git, 2019-11-06:
The problem can be possibly reproduced on STM32 as follows:
`static constexpr GPTConfig gpt5cfg = { /*!!!!!!!!*/100000, /* 1 MHz -> 100kHz timer clock.*/`

andreika-git, 2019-11-06:
Here's an example of only 1 injection event passed (see Inj1):
[image]

- "sch_inj PB4" is calling two engine->executor.scheduleForLater() for simult. injection inside handleFuelInjectionEvent() - skipped rarely (only twice on the screenshot);
- "crank" is 60-2 trigger signal;
- "hwTimerCallb" is hwTimerCallback() call - skipped frequently;
- "setHardwareU" is setHardwareUsTimer() call - skipped frequently;

rusefi, 2019-11-10:
enable self_stimulation
rpm 100
set cranking_rpm 200

rusefi, 2019-11-11:
Looks like this is all expected behavior - current implementation require the HW timer frequency to be exactly 1MHz, not more not less

```*
 * The specific 1MHz frequency is important here since 'setHardwareUsTimer' method takes microsecond parameter
 * For any arbitrary frequency to work we would need an additional layer of conversion.
 */
static constexpr GPTConfig gpt5cfg = { 1000000, /* 1 MHz timer clock.*/
		hwTimerCallback, /* Timer callback.*/
0, 0 };```

commit 41039159 (2019-11-11): Strange INJ1 signal behavior (when the frequency of scheduler's timer is less than 1MHz) fix #996
closed 2019-11-11 by rusefi