rusefi/rusefi #1041: injection scheduling artifact

rusefi, 2019-12-03:
As of today we have different scheduling mechanisms between spark and fuel.

Spark (and aux) uses a better scheduling approach where each event is in charge of scheduling the next one

Fuel still uses per-cycle scheduling approach which looks funny in transitional states. On the one hand, we do not care since that's fuel - on the other hand we know how to improve things and make things more consistent.

rusefi, 2019-12-03:
[image]

 See section 33.3s, 39.3s, 44.1s, 48.2s,51.4s etc.

[Injector issue.zip](https://github.com/rusefi/rusefi/files/3917924/Injector.issue.zip)

rusefi, 2019-12-03:
Maybe relevant maybe not video https://www.youtube.com/watch?v=oeq7oF9QIf0

rusefi, 2019-12-03:
See recent commit on #460 for how similar thing was improved for aux valuve

mck1117, 2020-07-28:
this is maybe a dupe of #1594

mck1117, 2022-06-23:
yeah, this is #1594

closed 2022-06-23 by mck1117