rusefi/rusefi #1051: Fried coil (?) on incompatible firmware change

rusefi, 2019-12-11:
updated firmware while connected to vehicle in violation of user manual

```2019-12-11_01_09_43_131: EngineState: FATAL error: Invalid ADC setting AUXF#1
2019-12-11_01_09_43_309: EngineState: FATAL error: Invalid ADC setting AUXF#1
2019-12-11_01_09_43_510: EngineState: FATAL error: Invalid ADC setting AUXF#1
2019-12-11_01_09_43_708: EngineState: FATAL error: Invalid ADC setting AUXF#1
```
because AUFX#1 bytes of configuration had something from previous version of ETB settings

saw magic smoke from front ignition coil

rusefi, 2019-12-11:
```2019-12-11_01_12_50_508: EngineState: FATAL error: Invalid ADC setting AUXF#1
2019-12-11_01_12_50_510: EngineState: confirmation_pins:4
2019-12-11_01_12_50_511: EngineState: pin PA0: adc input
2019-12-11_01_12_50_513: EngineState: pin PA2: adc input
2019-12-11_01_12_50_514: EngineState: pin PA4: adc input
2019-12-11_01_12_50_516: EngineState: pin PA5: trigger
2019-12-11_01_12_50_517: EngineState: pin PA6: adc input
2019-12-11_01_12_50_518: EngineState: pin PA10: USB ID
2019-12-11_01_12_50_520: EngineState: pin PA11: USB DM
2019-12-11_01_12_50_521: EngineState: pin PA12: USB DP
2019-12-11_01_12_50_522: EngineState: pin PB0: hip CS
2019-12-11_01_12_50_523: EngineState: pin PB3: SPI clock
2019-12-11_01_12_50_524: EngineState: pin PB4: SPI master in 
2019-12-11_01_12_50_526: EngineState: pin PB5: SPI master out
2019-12-11_01_12_50_527: EngineState: pin PB6: CAN TX
2019-12-11_01_12_50_532: EngineState: pin PB7: injector4
2019-12-11_01_12_50_533: EngineState: pin PB8: injector3
2019-12-11_01_12_50_533: EngineState: pin PB9: injector1
2019-12-11_01_12_50_534: EngineState: pin PB10: lcd DB7
2019-12-11_01_12_50_544: EngineState: pin PB11: hip int/hold
2019-12-11_01_12_50_544: EngineState: pin PB12: CAN RX
2019-12-11_01_12_50_546: EngineState: pin PC0: adc input
2019-12-11_01_12_50_546: EngineState: pin PC1: adc input
2019-12-11_01_12_50_547: EngineState: pin PC2: adc input
2019-12-11_01_12_50_547: EngineState: pin PC3: adc input
2019-12-11_01_12_50_551: EngineState: pin PC4: adc input
2019-12-11_01_12_50_551: EngineState: pin PC6: trigger
2019-12-11_01_12_50_552: EngineState: pin PC8: joy center
2019-12-11_01_12_50_554: EngineState: pin PC9: coil3
2019-12-11_01_12_50_562: EngineState: pin PD4: spi CS5
2019-12-11_01_12_50_564: EngineState: pin PD5: MalfunctionIndicator
2019-12-11_01_12_50_564: EngineState: pin PD7: main relay
2019-12-11_01_12_50_565: EngineState: pin PD10: joy A
2019-12-11_01_12_50_566: EngineState: pin PD11: joy D
2019-12-11_01_12_50_569: EngineState: pin PD12: led: running status
2019-12-11_01_12_50_569: EngineState: pin PD13: led: warning status
2019-12-11_01_12_50_570: EngineState: pin PD14: led: ERROR status
2019-12-11_01_12_50_570: EngineState: pin PD15: led: comm status
2019-12-11_01_12_50_572: EngineState: pin PE2: injector2
2019-12-11_01_12_50_573: EngineState: pin PE4: fuel pump relay
2019-12-11_01_12_50_573: EngineState: pin PE5: fan relay
2019-12-11_01_12_50_573: EngineState: pin PE7: lcd RS
2019-12-11_01_12_50_576: EngineState: pin PE8: dizzy tach
2019-12-11_01_12_50_576: EngineState: pin PE9: lcd E
2019-12-11_01_12_50_577: EngineState: pin PE11: lcd DB4
2019-12-11_01_12_50_577: EngineState: pin PE13: lcd DB5
2019-12-11_01_12_50_578: EngineState: pin PE14: coil1
2019-12-11_01_12_50_581: EngineState: pin PE15: lcd DB6
```

rusefi, 2019-12-11:
[image]

rusefi, 2019-12-11:
stm32f407 image 
[issue1051.zip](https://github.com/rusefi/rusefi/files/3948756/issue1051.zip)

rusefi, 2019-12-11:
https://www.youtube.com/watch?v=6E0d0K-BZQE

rusefi, 2019-12-11:
GIT_HASH "a44a0cad99fc64dce1428ce68e2e4a660ca2e60f"

ELF file:

[rusefi.zip](https://github.com/rusefi/rusefi/files/3952867/rusefi.zip)

rusefi, 2019-12-13:
It could be that me using old 0.4 Frankenso is part of the problem here

with newer boards we have
"117) RESOLVED IN R0.5 Add pull down and series resistors to hi/low mosfet driver"

mck1117, 2019-12-13:
uhh, that's a pretty dangerous fix.  Other boards use those pins - and might have stuff connected to them that causes misbehavior!

rusefi, 2019-12-13:
absolutely it's a questionable fix, but only Frankenso is the true and real board :)
what specific scenarios do we see where this would be an issue?

mck1117, 2019-12-13:
this also doesn't necessarily fix it - there's a period of time before chibios does its init work during which the pins are all forced to input (no pull up/down), during which no software has any control.  By the way, it is possible to bodge on those pulldown resistors on a Frankenso 0.4.  A 0603 or 0805 fits perfectly on top of the pins of the SOIC8, and there is a ground pin between the inputs.

rusefi, 2019-12-13:
the before-ChibiOS is quick and it's the lengthy power-on which fries the coils

great idea for Frankenso 0.4 HW fix! let me try that and if it works I will undo the nasty SW change. Re-opening ticket.

mck1117, 2019-12-13:
Yes, it might fix the burning coil, but it could still be long enough to charge and fire the coil (also bad).

mck1117, 2019-12-14:
fixed with hardware?

rusefi, 2019-12-14:
hardware solution

[image]

reverting questionable software solution

closed 2019-12-14 by rusefi

mck1117, 2019-12-14:
Sweet - that's basically what my board looks like too.

commit 1eeb92b2 (2019-12-13): Fried coil (?) on incompatible firmware change fix #1051
commit fe1d23fd (2019-12-14): Revert "Fried coil (?) on incompatible firmware change fix #1051"

This reverts commit 1eeb92b