rusefi/rusefi #1076: functional testing should not be that darn slow

rusefi, 2019-12-22:
at the moment functional testing on both simulator and real hardware takes too darn long, there should be a better way!

rusefi, 2019-12-22:
```2019-12-21_16_57_23_623<EOT>: Done in 677secs
2019-12-21_16_57_23_623<EOT>: +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
2019-12-21_16_57_23_623<EOT>: ++++++++++++++++++++++++++++++++++++  Real Hardware Test Passed +++++++++++++++
2019-12-21_16_57_23_623<EOT>: +++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++```

chuckwagoncomputing, 2023-01-30:
A random HW CI run I checked took 8m53s, and the "Run Hardware CI" step took 7m32.
I don't think it's reasonable to expect much more improvement.

closed 2023-01-30 by chuckwagoncomputing